verilator/verilator #3370: undefined symbol: _ZN16VerilatedContextD1Ev

margej, 2022-04-08:
Sorry if this is a dumb question. 

I'm trying to migrate from 4.102 to 4.202. I have read that VerilatedContext is a new feature in 4.2 

My existing 4.102 flow is to create a Verilated model of my core using --protect-lib, and include this in a VCS simulation. This is the only Verilated model in my environment, so I assume I need only a default singular VerilatedContext somewhere.

When I compile the model with 4.202 (the most recent version I am able to compile on my ancient CentOS7 crate) I see the above error when I run the executable.

./simv: symbol lookup error: verilator/libverilated_thing.so: undefined symbol: _ZN16VerilatedContextD1Ev

The documentation mentions a default context being created by default, but I image this is not practical for a protect-lib model, so my question is, what do I need to add to my simulation executable to expose this symbol in a usable way?

Do I just need to export a function called VerilatedContext which returns a pointer to a lump of memory my model can use to store its state?

Apologies if this is already documented, but I couldn't find it.

wsnyder, 2022-04-09:
It's complaining about:

          echo '_ZN16VerilatedContextD1Ev' | c++filt
          VerilatedContext::~VerilatedContext()

That is defined in verilated.cpp, so I suspect you either didn't properly include that in your library, or didn't link with the appropriate object, or perhaps linked against an old object.  Any case, something to do with your makefiles etc.

margej, 2022-04-11:
Thanks,
Adding ${VERILATOR_ROOT}/include/verilated.cpp to the verilator command line resolved the issue.
I didn't need to do this before so it might be worth adding a note to the new chapter in the docs on VerilatedContext ?

wsnyder, 2022-04-11:
I don't think that is related to VerilatedContext, I suspect it's related to your linker/Makefile rules not linking against the proper runtime libraries. Adding it to the command line is just adding it to different make variables which you are presumably using in your makefile, versus others you are not (VM_SUPPORT_FAST/VM_SUPPORT_SLOW) .  Are you seeing that problem using the --exe option with Verilator where you use the makefile it generated rather than your own?

margej, 2022-04-11:
Ah, I've found the problem.
I use the verilator generated makefile, BUT, I find I have to patch it as follows to get user objects included in the .so 

```
#	@echo Patch ${MDIR}/V${TOPMOD}.mk to include user objects
#	@sed -i 's|$$(VK_GLOBAL_OBJS)|$$(VK_BLOBAL_OBJS) $$(VK_USER_OBJS)|g' ${MDIR}/V${TOPMOD}.mk
```

The typo in the above meant I was erroneously excluding VK_GLOBAL_OBJS (I.e. verilated.o) from my library, which I guess never used to matter, but the addition of VerilatedContext means symbols in verilated.o are now required by my simulation.

Fixing the above typo resolves this issue. Thanks for the help.

Is it worth raising a PR to add VK_USER_OBJS to the generated makefile, or is it only missing because I'm doing something else wrong?

wsnyder, 2022-04-11:
I think you're patching this code:

```
- $(VK_GLOBAL_OBJS): %.o: %.cpp
-        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_GLOBAL) -c -o $@ $<
+ $(VK_GLOBAL_OBJS) $(VK_USER_OBJS): %.o: %.cpp
+        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_GLOBAL) -c -o $@ $<
```

I think that makes sense to fix, would you mind making a pull request?

wsnyder, 2022-04-11:
The fix you suggested is (I think) different from what I thought above.  We don't want the globals in a generated library .a/.so file because there can be multiple .a/.so's linked into the same design.   Perhaps we should also make a verilated.a/.so and require the user to link against that too.

That is if doing 3 designs you'd link design1.a design2.a verilated.a.

gezalore, 2022-04-11:
I'm confused. The "globals" (i.e.: verilated*.cpp) were already in that library (based on the changed line), the PR adds the user objects. Also, your 'fix' above was to a compilation rule, so should have no effect on linking.

margej, 2022-04-11:
I agree, but VK_GLOBAL_OBJS are already included for libCreate, My change is just adding VK_USER_OBJS ?

wsnyder, 2022-04-11:
Right, I misread it, please see comments in pull request.

wsnyder, 2022-04-12:
That is, I think there's confusion because it doesn't match what's in master.

wsnyder, 2022-04-12:
Thanks for fixing.

closed 2022-04-12 by wsnyder