verilator/verilator #3374: Wrong result when accessing array

notlqr, 2022-04-11:
## Bug Description
An example: 
`main.cpp`:
```c++
#include "VTop.h"
#include "verilated.h"

int main() {
        VerilatedContext* contextp = new VerilatedContext;
        VTop* top = new VTop{contextp};
        top->clock = 0;
        top->index = 0;
        top->eval();
        top->clock = 1;
        top->eval();
        for (int i = 0; i <= 7; i++) {
                printf("i is %d,", i);
                top->clock = 0;
                top->index = i;
                top->eval();
                top->clock = 1;
                top->eval();
                printf("Res is %x\n", top->out);
        }
        top->final();
        delete top;
}
```

`Top.v`:
```verilog
module Top(
        input clock,
        input [2:0] index,
        output [27:0] out
        );
        reg [27:0] res;

        wire [2:0] flipped_index = ~index;
        always @(posedge clock) begin
                automatic logic [7:0][26:0] array = {27'h33, 27'h34, 27'h35, 27'h36, 27'h37, 27'h38, 27'h39, 27'h40};
                res <= array[~flipped_index];
        end

        assign out = res;
endmodule
```

Build command: `verilator -Wall -Wno-WIDTH --cc -O0 --exe --build main.cpp Top.v`

It seems that 2 factors are necessary to trigger this bug: 
- the width of array elements should not be power of 2 (eg. 32)
- the index must be flipped when accessing the array

## Verilator version:
Verilator 4.221 devel rev v4.220-21-g5f333be9 (mod)

## OS:
Arch Linux with verilator master and Ubuntu 20.04LTS with verilator 4.218

wsnyder, 2022-04-13:
I get WIDTH warnings which indicates the problem in in V3Width.

Would you be willing to take a look into fixing this?

notlqr, 2022-04-13:
I can take a look. Where should I start from?

wsnyder, 2022-04-13:
1. Create a test, an example based on yours is attached.
2. Run this with --debug, and perhaps also a similar case that uses index to avoid the bug with --debug, compare the obj_dir/*width*tree files to see what's going on.
3. --debugi-V3Width 9 might also help
4. I suspect something in determining the width of the array is not going correctly.
The data type determination is step-by-step, so probably it is sizing the array index that is incorrect.  Then the false WIDTH warning is because the inversion doesn't match the (wrong) type of the array index.

[bug3374.patch.txt](https://github.com/verilator/verilator/files/8481111/bug3374.patch.txt)

leonardt, 2023-03-16:
I ran into what I think may be a related issue.

Verilog
```verilog
module Foo(
  input  [7:0] I,
  output [7:0] O);

  wire [1:0][7:0] _GEN = {{I - 8'h1}, {I + 8'h1}};
  assign O = _GEN[I != 8'hFF & I[0]];
endmodule
```

lint result
```
❯ verilator --lint-only Foo.v
%Warning-WIDTH: Foo.v:7:30: Operator AND expects 32 or 4 bits on the LHS, but LHS's NEQ generates 1 bits.
                          : ... In instance Foo
    7 |   assign O = _GEN[I != 8'hFF & I[0]];
      |                              ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: Foo.v:7:30: Operator AND expects 32 or 4 bits on the RHS, but RHS's SEL generates 1 bits.
                          : ... In instance Foo
    7 |   assign O = _GEN[I != 8'hFF & I[0]];
      |                              ^
%Error: Exiting due to 2 warning(s)
```

Is there anyone actively looking at this? I'd be happy to help, but could use pointers to relevant places in the code that might be related.

wsnyder, 2023-03-16:
>I != 8'hFF & I[0]

Means 
>I != (8'hFF & I[0])

And verilator is correctly warning you.

I suspect you wanted

>I != 8'hFF && I[0]

leonardt, 2023-03-17:
I initially thought the same thing, but then I checked the system verilog spec: https://ieeexplore.ieee.org/document/8299595

Page 257, Table 11-2

It turns out that `!=` has higher precedence than `&` (and `&&` for that matter)

leonardt, 2023-03-17:
I realized I should check the verilog spec because maybe there was an inconsistency, but similar precedence ordering there: https://ieeexplore.ieee.org/document/1620780

wsnyder, 2023-03-17:
Higher means it binds more tightly, which matches my example.  Try edaplayground.com if you want to experiment.

wsnyder, 2023-03-17:
I realized you're still right about & vs && being same, let me add parens and look into this more.

wsnyder, 2023-03-17:
@leonardt your issue is different, I get the correct results under the cases I tried after disabling the WIDTH warning.  Can you please file a new bug?

paul-demo, 2025-06-10:
Original issue is not reproducible at mainline. Here's a test-regress target. It passes for both 27-bit widths everywhere and for the case original poster had which is that a 27-bit slice gets stored in a 28-bit slice. You have to waive WIDTHEXPAND to simulate the latter case.

```systemverilog
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2017 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

// Required because original issue catches 27-bit values with 28-bit variables
/* verilator lint_off WIDTHEXPAND */

module t(/*AUTOARG*/
   // Inputs
   clk
);
  input clk;

  logic [2:0] index;
  logic [27:0] out;

  parameter int NUM_ITERS = 16;
  int cyc = 0;
  int num_correct = 0;

  assign index = 3'(cyc);

  dut u_dut(.*);

  always @ (posedge clk) begin
    cyc <= cyc + 1;

    if (cyc >= 1) begin
      if (out == 28'(((cyc - 1) % 8) + 1)) begin
        num_correct <= num_correct + 1;
      end else begin
        $stop;
      end
    end

    if (cyc > NUM_ITERS) begin
      if (num_correct == NUM_ITERS) begin
        $display("num_correct = %d", num_correct);
        $write("*-* All Finished *-*\n");
        $finish;
      end else begin
        $stop;
      end
    end
  end

endmodule

module dut(
  input clk,
  input [2:0] index,
  output [27:0] out
);
  reg [27:0] res;
  wire [2:0] flipped_index = ~index;

  always @(posedge clk) begin
    automatic logic [7:0][26:0] array = {27'h08, 27'h07, 27'h06, 27'h05, 27'h04, 27'h03, 27'h02, 27'h01};
    res <= array[~flipped_index];
  end

  assign out = res;
endmodule
```

paul-demo, 2025-06-10:
```text
num_correct =          16
*-* All Finished *-*
- t/t_arraysel_logic_operand.v:42: Verilog $finish
vlt/t_arraysel_logic_operand: Self PASSED
==SUMMARY: Passed 1  Failed 0  Time 0:01
```

wsnyder, 2025-06-10:
Thanks for checking this, assuming works now.

closed 2025-06-10 by wsnyder