verilator/verilator #3378: Hierarchical compile fails with structs as IO

abond-jt, 2022-04-11:
When compiling with --hierarchical modules marked as /*verilator hierarchical_blocks*/ can't have IO ports that are of type struct.

Code Example:
`
typedef struct packed {
  logic        vld;
  logic [31:0] count;
} sts_t;

module submodule (
  input wire logic       clk_i,
  input wire logic       rst_n_i,
  output var sts_t       sts_o
); /*verilator hier_block*/ 

  logic[31:0] counter_q;

  always_ff @(posedge clk_i or negedge rst_n_i) begin
    if (!rst_n_i) begin
      counter_q <= 32'b0;
    end
    else begin
      counter_q <= counter_q + 1;

      $display("%m: Counter = 0x%x", counter_q);

      if (counter_q >= 32'hffff)
        $finish;
    end
  end

  always_comb sts_o.vld   = |counter_q;
  always_comb sts_o.count =  counter_q;

endmodule

module dut (
  input wire logic       clk_i,
  input wire logic       rst_n_i
);

  submodule s_u (.clk_i(clk_i),
                 .rst_n_i(rst_n_i),
                 .sts_o()
                );
endmodule
`

`
#ifdef ENV_SIM_VERILATOR
#include <verilated.h>
#include "Vdut.h"
#include "Vdut___024root.h"
#ifdef WAVES
#include "verilated_vcd_c.h"
#endif

int main(int argc, char** argv, char** env) {
    // Prevent unused variable warnings
    if (false && argc && argv && env) {}
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};

    Vdut*             dut  = new Vdut("dut");

#ifdef WAVES
    Verilated::traceEverOn(true);
    VerilatedVcdC* tfp = new VerilatedVcdC;
    dut->trace(tfp, 99);  // Trace 99 levels of hierarchy (or see below)
    tfp->open("waves.vcd");
#endif

    dut->rootp->rst_n_i = 0;
    dut->rootp->clk_i   = 0;

    // Simulate until $finish
    while (!Verilated::gotFinish()) {

        contextp->timeInc(1);

        dut->rootp->clk_i = !dut->clk_i;

        if (!dut->rootp->clk_i) {
            if (contextp->time() > 1 && contextp->time() < 10) {
                dut->rootp->rst_n_i = !1;  // Assert reset
            } else {
                dut->rootp->rst_n_i = !0;  // Deassert reset
            }
        }
        // Evaluate model
        dut->eval();
#ifdef WAVES
        tfp->dump(contextp->time());
#endif
    }

    // Final model cleanup
    dut->final();

    // Destroy model
    delete dut;

    // Return good completion status
    return 0;
}
#endif
`

`
non_hier: dut.sv tb.cpp
	verilator --exe -cc -CFLAGS "-DENV_SIM_VERILATOR" -timescale 1ns/1ps dut.sv tb.cpp
	make -C obj_dir/ -f Vdut.mk
	./obj_dir/Vdut

hier: dut.sv tb.cpp
	verilator --exe -cc -hierarchical -CFLAGS "-DENV_SIM_VERILATOR" -timescale 1ns/1ps dut.sv tb.cpp
	make -C obj_dir/ -f Vdut.mk
	./obj_dir/Vdut

clean:
	rm -rf obj_dir
`

wsnyder, 2022-04-11:
Confirmed.  Perhaps you'd be willing to look into making a fix for this?  I think it could just convert the struct to a fixed-size non-struct when generating the Verilog wrapper code.

wsnyder, 2022-04-12:
Make test_regress files similar to your example above  (you probably don't need the .cpp file and can use the standard one that other tests use).
Use --debug to see the structure of the AST nodes.
V3ProtectLib.cpp's handleInput currently just copies the variable, this should probably instead change packed structures to non-structures.

@yTakatsukasa might have some other suggestions too.

yTakatsukasa, 2022-04-12:
Hierarchical verilation fully relies on `--lib-create` (previously `--protect-lib`).
Once `--lib-create` supports packed struct, `--hierarchical` also supports it.
I'd suggest developing/debugging using `--lib-create` against submodule that has struct in/out because you can launch verilator directly with the option while `--hierarchical` calls child verilator.

wsnyder, 2022-04-14:
All packed structures are equivalent (numerically) to something of the structure's dtypep->width().   To look for a structure you'd use VN_IS(..., AstNodeUOrStructDType) then check for packed().  Please also test packed unions as should work likewise.

PR #3892 (2023-01-23): Allow struct I/O for --lib-create
fixes #3378

I don't really understand the codebase and this was done through pattern-matching, so feel free to point out deficiencies.
closed 2023-01-23 by wsnyder