verilator/verilator #3379: $readmemh: Unsupported: Blocked and non-blocking assignments to same variable

solomatnikov, 2022-04-12:
`$readmemh` and blocking assignment to an associative array produces an error:

```
%Error-BLKANDNBLK: dv/testmem.sv:15:27: Unsupported: Blocked and non-blocking assignments to same variable: 'top.bench.testmem.mem'
   15 |    logic [DATA_WIDTH-1:0] mem [logic [ADDR_WIDTH-1:0]];
      |                           ^~~
                   top.v:55:46: ... Location of blocking assignment
   55 |        $readmemh(file, bench.testmem.mem);
      |                                              ^~~
                   dv/testmem.sv:28:6: ... Location of nonblocking assignment
   28 |      mem[w_addr][i*8+7:i*8] <= w_data[i*8+7:i*8];
      |      ^~~
                   ... For error description see https://verilator.org/warn/BLKANDNBLK?v=4.221
%Error: Exiting due to 1 error(s)
```

Even though `test_regress/t/t_altera_lpm.v` has similar code for simple array:

```
        if (lpm_file != "UNUSED")
        begin
            mem.convert_to_ver_file(lpm_file, lpm_width, ram_initf);
            $readmemh(ram_initf, mem_data);
        end

        tmp_q = ZEROS;
    end

// ALWAYS CONSTRUCT BLOCK                                                                                                                                                 
    always @(posedge i_inclock)
    begin
        if (lpm_address_control == "REGISTERED")
        begin
            if ((we) && (use_eab != "ON") &&
                (lpm_hint != "USE_EAB=ON"))
            begin
                if (lpm_indata == "REGISTERED")
                    mem_data[address] <= data;
                else
                    mem_data[address] <= pdata;
            end
```

Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

**Yes, an example can be done**

What 'verilator --version' are you using?

**Verilator 4.221 devel rev v4.220-21-g5f333be9**

Did you try it with the git master version?

**Yes**

What OS and distribution are you using?

**Linux 4.18.0-373.el8.x86_64 #1 SMP Tue Mar 22 15:11:47 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux**

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2022-04-12:
You can disable the error with the usual lint_off to get around this.

It probably makes more sense that Verilator automatically disable this check under $readmem/$writemem, as it seems unlikely that the user is unaware of what they are doing.  Would you be willing to make a pull request including test to do that?

solomatnikov, 2022-04-19:
Unfortunately, with simple test case this doesn't happen:

```
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); $stop; end while(0);
`define checks(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='%s' exp='%s'\n", `__FILE__,`__LINE__, (gotv), (expv)); $stop; end while(0);
`define checkg(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='%g' exp='%g'\n", `__FILE__,`__LINE__, (gotv), (expv)); $stop; end while(0);

module sub (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   integer cyc = 0;

   // associative array of an associative array
   logic [31:0] a [logic [31:0]];

   always @ (posedge clk) begin
      cyc <= cyc + 1;
      if (cyc == 1) begin
         a[5] <= a[5] + 1;
         a[6] <= a[6] + 1;
      end
      else if (cyc == 2) begin
         `checkh(a[5], 6);
         `checkh(a[6], 7);
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule

module t (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   sub sub ( .* );

   initial begin
      $readmemh("t/t_sys_readmem_h2.mem", sub.a);
   end
endmodule // t
```

closed 2022-06-04 by wsnyder