verilator/verilator #3383: Version 5 development branch

wsnyder, 2022-04-13:
I've created the develop-v5 branch.  This will collect changes towards v5.002.

There are two current candidates for inclusion here:
* timed coroutines (#3363)
* improved scheduling (#3329)

We welcome people to use this branch to perform alpha testing.

wsnyder, 2022-05-15:
@gezalore has merged the first large improvement which has a compliant scheduler #3329 no more misgenerated clocks!

We'd appreciate beta testers; please try the develop-v5 branch, and if you have issues file github issues as usual.  Thanks.

gezalore, 2022-05-15:
@dpetrisko might be interested?

Alex-Mann, 2022-09-15:
Just a note that these lines are essentially the same in `develop-v5` branch:
https://github.com/verilator/verilator/blob/develop-v5/bin/verilator#L344
https://github.com/verilator/verilator/blob/develop-v5/bin/verilator#L347

wsnyder, 2022-09-16:
@Alex-Mann thanks I fixed the --main.

wsnyder, 2022-10-02:
@gezalore yesterday merged develop-v5 branch into master!   Feedback is welcome to prepare this to become the next stable release.

closed 2022-10-15 by wsnyder