verilator/verilator #3385: ref task parameter doesn't work the same as VCS

mytbk, 2022-04-14:
In the following example, when INOUT is ref, verilator displays "flag=0" in the second line, while VCS 2017.03 displays "flag=1".

```
//`define INOUT inout
`define INOUT ref

module automatic top;

  task update(`INOUT bit flag, input bit upflag);
    flag = upflag? 1: flag;
  endtask: update

  bit my_flag;
initial begin
  update(my_flag, 1);
  $display("flag=%d", my_flag);
  update(my_flag, 0);
  $display("flag=%d", my_flag);
  $finish();
end
endmodule;
```

Verilator version: Verilator 4.221 devel rev v4.220-24-gbaff64a4

wsnyder, 2022-04-14:
Confirmed.  Perhaps you'd be willing to try a pull request to fix this?

It looks like V3Task.cpp is zeroing  __Vtask_t__DOT__update__1__upflag on entry to the function (because it's a bit), but shouldn't do that for refs.

mytbk, 2022-04-14:
When I use -O3, the generated code is:

```
VL_ATTR_COLD void Vtop___024root___initial__TOP__0(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___initial__TOP__0\n"); );
    // Init
    CData/*0:0*/ top__DOT__my_flag;
    CData/*0:0*/ __Vtask_top__DOT__update__1__flag;
    // Body
    VL_WRITEF("flag=1\n");
    top__DOT__my_flag = __Vtask_top__DOT__update__1__flag;
    VL_WRITEF("flag=%1#\n",1,top__DOT__my_flag);
    VL_FINISH_MT("top.sv", 16, "");
}
```

With -O0, I get:

```
VL_ATTR_COLD void Vtop_top___initial__TOP__top__0(Vtop_top* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+  Vtop_top___initial__TOP__top__0\n"); );
    // Init
    CData/*31:0*/ __Vtemp_hac07d98c__0;
    CData/*31:0*/ __Vtemp_h0043aa71__0;
    // Body
    vlSelf->__Vtask_update__0__upflag = 1U;
    __Vtemp_hac07d98c__0 = (1U & VL_BITSEL_IIII(32, 
                                                ((IData)(vlSelf->__Vtask_update__0__upflag)
                                                  ? 1U
                                                  : 
                                                 VL_EXTEND_II(32,1, (IData)(vlSelf->__Vtask_update__0__flag))), 0U));
    vlSelf->__Vtask_update__0__flag = __Vtemp_hac07d98c__0;
    vlSelf->__PVT__my_flag = vlSelf->__Vtask_update__0__flag;
    VL_WRITEF("flag=%1#\n",1,vlSelf->__PVT__my_flag);
    vlSelf->__Vtask_update__1__upflag = 0U;
    __Vtemp_h0043aa71__0 = (1U & VL_BITSEL_IIII(32, 
                                                ((IData)(vlSelf->__Vtask_update__1__upflag)
                                                  ? 1U
                                                  : 
                                                 VL_EXTEND_II(32,1, (IData)(vlSelf->__Vtask_update__1__flag))), 0U));
    vlSelf->__Vtask_update__1__flag = __Vtemp_h0043aa71__0;
    vlSelf->__PVT__my_flag = vlSelf->__Vtask_update__1__flag;
    VL_WRITEF("flag=%1#\n",1,vlSelf->__PVT__my_flag);
    VL_FINISH_MT("top.sv", 16, "");
}
```

Both -O0 and -O3 inline the task update(). In the -O3 code, it has a ``__Vtask_top__DOT__update__1__flag`` variable, which is not initialized. In the -O0 code, it has two independent, initialized as zero, class members ``__Vtask_update__0__flag`` and ``__Vtask_update__1__flag``. So I think verilator tries to instantiate a new variable on each call to the task, but it can be wrong for ref parameters.

wsnyder, 2022-04-25:
I suggest changing the test to specifically look for a ref.  Then would you perhaps be willing to convert this to a pull request, including a test please?

wsnyder, 2023-09-15:
Unfortunately note I think we have to support ref as a C++ ref rather than copying as inout (e.g. #4489 first try).
1. As you note, forks won't work, and UVM uses forks and ref.
2. The t_queue_persistence test I just added, which is based on the HRM/sv-tests, needs it (similar to what UVM does).
3. UVM/others use ref to pass huge structures (e.g. register map) of tens of megabytes, and copying them would greatly harm performance.

RRozak, 2023-09-18:
Verilator currently handles inout arguments as C++ ref. #4489 makes ref arguments being handled in the same way as inout. It isn't 100% correct, but I think it works incorrectly only in forks (I am not sure about it).
Generally I think that #4489 handles ref arguments correctly. The problem with inout arguments will remain, but new types will be supported.

RRozak, 2023-09-18:
As I see, the copying of `inout` arguments was recently added: https://github.com/verilator/verilator/commit/ab13548018e3db957920bdcdc593358602ca87d9.
In order to support `ref` arguments correctly, I need to handle them as inout arguments were handled previously. I will update #4489.

wsnyder, 2024-11-25:
Self-checking test:

```
`define INOUT ref

module automatic t;

   task update(`INOUT int flag, input bit upflag);
      flag = upflag ? 1 + flag : flag;
   endtask

   int my_flag;

   initial begin
      update(my_flag, 1);
      $display("flag=%0d", my_flag);
      if (my_flag !== 1) $stop;
      update(my_flag, 0);
      $display("flag=%0d", my_flag);
      if (my_flag !== 1) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule

```

wsnyder, 2024-11-26:
Believe fixed earlier, test cases added and pass.