verilator/verilator #3387: with operator and type casting

xiak95, 2022-04-16:
I think the following is a correct usage of with operator to cast type for queue sum

```
    /* verilator lint_off WIDTH */
    byte myqueue [$] = {2,-1,127};
    int total;
    initial begin
        total = myqueue.sum() with (int'(item));
        $display("%d",total );
    end
    /* verilator lint_on WIDTH */
```
however its crashing Verilator 4.221 devel rev v4.220-25-g7bfc1a00 compiled from master branch of this repo

```
verilator -Wall --trace -cc testbench.sv e1/DUT.sv --exe main.cpp --top-module testbench -CFLAGS "-DTIMEOUT=100" -Wno-UNUSED
%Error: Internal Error: e1/DUT.sv:42:31: ../V3LinkDot.cpp:397: Module/etc never assigned a symbol entry?
   42 |         total = myqueue.sum() with (int'(item));
      |                               ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make: *** [Makefile:6: obj_dir/Vtestbench] Error 1
```

wsnyder, 2022-04-27:
Confirmed.  This is probably something relatively simple going wrong in V3LinkDot as this syntax likely hadn't been tested before now.  Perhaps you'd be willing to investigate further?  Basically you'd need to look at why V3LinkDot isn't iterating into that construct and the assertion is firing.

wsnyder, 2022-05-15:
There was two bugs fixed, first the crash, second the WIDTH warning suppression shouldn't be needed - that indicated wrong internal sizing.

closed 2022-05-15 by wsnyder