verilator/verilator #3389: What is the purpose of cleaning up the model? (`delete top`)

goekce, 2022-04-17:
I am new to the Verilator and trying to create a minimal C++ model driver for me and my students. While browsing the examples I noticed the following clean up routines:

https://github.com/verilator/verilator/blob/7bfc1a00a723cd91d8d52c146b5e7b45844a2ed6/examples/make_hello_c/sim_main.cpp#L34-L41

https://github.com/verilator/verilator/blob/7bfc1a00a723cd91d8d52c146b5e7b45844a2ed6/examples/make_tracing_c/sim_main.cpp#L112-L124

From https://github.com/verilator/verilator/blob/master/docs/guide/example_cc.rst:

```
....
delete top;
delete contextp;
return 0;
```
```
I was pondering
- `delete top`
- `delete contextp`
- `return 0`.

In my experiments I saw that `delete top` was required to successfully write the FST trace file (otherwise Gtkwave could not open it). `delete contextp` did not have any effect. Regarding `return 0` I found [this reference](https://stackoverflow.com/a/204483) from which I understand that `return 0` can be omitted.

- What is the purpose of these statements?
- If they should not be omitted, why?

<details><summary>If needed here is my testbench driver</summary>
```cpp
#include "Vmajority_of_five_tb.h"
#include "verilated.h"
int main(int argc, char** argv, char** env) {
	VerilatedContext* contextp = new VerilatedContext;

	contextp->traceEverOn(true);
	// Compute traced signals

	contextp->commandArgs(argc, argv);
	Vmajority_of_five_tb* top = new Vmajority_of_five_tb{contextp};
	top->clk = 0;
	while (!contextp->gotFinish()) {
			contextp->timeInc(1);
			top->clk = !top->clk;
			top->eval();
	}
	top->final();
	delete top;
	// Must be deleted manually. Otherwise the signal trace is not written properly.
}
```

wsnyder, 2022-04-18:
In C++ a delete calls the destructor of an object, which is otherwise not done (in some cases), therefore Verilator doesn't know when to clean up and close traces.

BTW a better way is to construct it like this, which will automatically delete the object when appropriate.  (See the more complicated tracing examples.)

    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    ....
    const std::unique_ptr<Vtop> top{new Vtop{contextp.get(), "TOP"}};

closed 2022-04-18 by wsnyder

goekce, 2022-04-19:
Lastly: Do you think `return 0` is really needed?

Thanks for your explanations Wilson!

goekce, 2022-04-20:
I researched about `return 0` in `main`. It does not seem to be required:

From [C++ 2020 Standard draft 6.9.3.1](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2020/n4849.pdf#subsubsection.6.9.3.1):

```
(5) ...  If control flows off the end of the compound-statement of main, the effect is equivalent to a return with operand 0.
```

Using `return 0` is not an error. But I think a getting started document should contain minimal examples to not distract and demotivate new users.

goekce, 2022-04-20:
After some experiments I came up with this minimal example for a sequential design. If standard pointers are used, then `delete` seems to be necessary. Using `std::unique_ptr` can help. Another alternative is using static objects.

Additionally I introduced macros for computed include statements to specify the top design name once. These macros could be provided in Verilator:

```cpp
#define TOP majority_of_five_tb

#define vIncFile3(x) #x
#define vIncFile2(x) vIncFile3(V##x.h)
#define vIncFile(x) vIncFile2(x)
#include vIncFile(TOP)

#define VTOP3(x) V##x
#define VTOP2(x) VTOP3(x)
#define VTOP VTOP2(TOP)

#include "verilated.h"

int main(int argc, char** argv) {
	VerilatedContext context;
	context.traceEverOn(true);
	context.commandArgs(argc, argv);
	VTOP top(&context);

	top.clk = 0;
	while (!context.gotFinish()) {
			context.timeInc(1);
			top.clk = !top.clk;
			top.eval();
	}
}
```

`TOP` could also be specified in the command:

```makefile
TOP = majority_of_five_tb

default:
	verilator \
		--cc \
		--exe \
		--build \
		--trace-fst \
		-j \
		-CFLAGS -DTOP=$(TOP) \
	   	$(TOP).{v,cpp}
	obj_dir/V$(TOP) +trace

clean:
	-rm -rf obj_dir
```

Does this approach make sense?

wsnyder, 2022-04-25:
>-CFLAGS -DTOP=$(TOP)

Yes that makes sense to do and is similar to what the self-tests do.  Verilator supports having many models simultaneously so this isn't something I'd like Verilator to do automatically.

goekce, 2022-04-26:
> Verilator supports having many models simultaneously so this isn't something I'd like Verilator to do automatically.
Makes sense. For beginners a minimal example like above could be useful though.