verilator/verilator #3391: Verilog compile hangs with XMR to function inside genblk

VarunKoyyalagunta, 2022-04-19:
Failing test here -
https://github.com/VarunKoyyalagunta/verilator/commit/ce04c291bb093e35f7d181cf70db3ca0984bd0e2
```
$ ./test_regress/t/t_xmr_genblk_function.pl --debug --debugi-V3LinkDot 9
- V3LinkDot.cpp:673:      findSymPrefixed block__DOT__set under se0x5555562df000 as t__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set  at se0x5555562df000
- V3SymTable.h:151:        SymFind   se0x5555562df000 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562ded80 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562dece0 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562df000 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562ded80 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562dece0 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562df000 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562ded80 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562dece0 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
- V3SymTable.h:151:        SymFind   se0x5555562df000 't__DOT__test__DOT__subtest__DOT__block__DOT__block__DOT__set' -> NONE
[last message repeated forever]
```

verilator --version:
Tried `4.220` and master (`Verilator 4.221 devel rev v4.220-26-g880a9be3`)

OS:
Linux

I can work around this by creating an additional layer of module hierarchy inside the genblk that encapsulates the function.

I don't really have a lot of time to help out with this right now, but thought I'd report it and can look into fixing it later if it's still open.

yodalee, 2022-04-25:
I do some investigate and it seems that the code is blocked in function `findSymPrefixed`.
It is weird that the following code:
```cpp
        const string prefix = lookupSymp->symPrefix();
        VSymEnt* foundp = nullptr;
        while (!foundp) {
            foundp = lookupSymp->findIdFallback(prefix + dotname);  // Might be nullptr
            if (prefix.empty()) break;
            const string nextPrefix = removeLastInlineScope(prefix);
            if (prefix == nextPrefix) break;
        }
```
The `nextPrefix` is actually not used in anywhere. So in this case, the `foundp` is nullptr, `nextPrefix` is not equal to prefix, it goes into infinite loop.
Maybe we need a `prefix = nextPrefix` after checking `prefix == nextPrefix`. Just guessing by the variable name.

wsnyder, 2022-04-27:
Thanks for the contribution, fixed.

closed 2022-04-27 by wsnyder