verilator/verilator #3395: Side effect function inside unsatisfied if condition get executed

weisrc, 2022-04-22:
**Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)**

```sv
typedef struct {int pc} state_t;

function int A(inout state_t state);
  state.pc++;
  return 1;
endfunction

function int B(inout state_t state);
  state.pc++;
  return 2;
endfunction

module test(input logic clk);
  int value = 0;
  state_t state;
  always @(posedge clk) begin
    if (condition) value = A(state);
    else begin
      value = B(state);
      // $display("difference");
    end
    // or
    value = condition ? A(state) : B(state);
  end
endmodule
```
The code above will increment pc twice regardless of the condition. However when I uncomment `$display("difference")`, it works fixes: increments pc once. I am not sure if the code snippet provided above will reproduce the bug. But the code in my repo does.

If statement: https://github.com/weisrc/nesv/blob/main/src/cpu/matrix/1.sv#L7
Function 1: https://github.com/weisrc/nesv/blob/main/src/cpu/actions/addressing.sv#L66
Function 2:  https://github.com/weisrc/nesv/blob/main/src/cpu/actions/addressing.sv#L85
Output: https://github.com/weisrc/nesv/runs/6122372041?check_suite_focus=true#step:4:579

**With `$display("difference")`**
If statement: https://github.com/weisrc/nesv/blob/main/src/cpu/matrix/1.sv#L11
Output: https://github.com/weisrc/nesv/runs/6122595622?check_suite_focus=true#step:4:578

**What 'verilator --version' are you using?**  
Verilator 4.218 2022-01-17 rev UNKNOWN.REV
**Did you try it with the git master version?**
No
**What OS and distribution are you using?**
Arch Linux

**May we assist you in trying to fix this in Verilator yourself?**
Yes, please! (I am relatively new to verilog, so I don't think I can do much on my own.)

Thank you! Verilator is amazing!

yTakatsukasa, 2022-04-22:
If adding `-Os` hides the problem, I think this error comes from incorrect `AstNode::isPure()` used in 
https://github.com/verilator/verilator/blob/880a9be3b10763b1d0b68414c1fa02700b3377d2/src/V3Split.cpp#L426

This issue reminds me of potential problem of #2861, but not sure if it is the case.

yTakatsukasa, 2022-04-22:
Your snippet shows the wrong result.
My guess in the previous comment seems not the case.

I'd suggest make the verilator standard test case as in test_regress/t and run the added test with `--dump-tree` option when you run verilator.

obj_dir/Vt_036_deadDtypesScoped.tree may have something like below. This looks ok.
```
    1:2:2:2:2: ASSIGN 0x564001ec93f0 <e3201> {c17ba} @dt=0x564001ed3110@(sw32)                                                                                                                                                                                                    
    1:2:2:2:2:1: COND 0x564001ecd940 <e3199> {c17bc} @dt=0x564001ed3110@(sw32)    
    1:2:2:2:2:1:1: VARREF 0x564001ed7d30 <e3195> {c17aj} @dt=0x564001ed2710@(G/w1)  condition [RV] <- VARSCOPE 0x564001ed2320 <e4033> {c13bn} @dt=0x564001ed2710@(G/w1)  TOP->condition -> VAR 0x564001ecf140 <e1851> {c13bn} @dt=0x564001ed2710@(G/w1)  condition [PI] INPUT [P] [VSTATIC]  PORT
    1:2:2:2:2:1:2: FUNCREF 0x564001ed21f0 <e3196> {c17bc} @dt=0x564001ed3110@(sw32)  A pkg=0x564001ec81d0 -> FUNC 0x564001eda3b0 <e1322> {c3ao} @dt=0x564001ed3110@(sw32)  A
    1:2:2:2:2:1:2:3: ARG 0x564001ed5b50 <e140> {c17be}                                                                                                                          
    1:2:2:2:2:1:2:3:1: VARREF 0x564001ed7ff0 <e631> {c17be} @dt=0x564001ed34e0@(w32)  t__DOT__state [LV] => VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:1:3: FUNCREF 0x564001ed4e90 <e3197> {c19ar} @dt=0x564001ed3110@(sw32)  B pkg=0x564001ec81d0 -> FUNC 0x564001ecc290 <e4020> {c8ao} @dt=0x564001ed3110@(sw32)  B
    1:2:2:2:2:1:3:3: ARG 0x564001ed4db0 <e149> {c19at}                                                                                                                          
    1:2:2:2:2:1:3:3:1: VARREF 0x564001ed65b0 <e675> {c19at} @dt=0x564001ed34e0@(w32)  t__DOT__state [LV] => VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]
```

obj_dir/Vt_039_task.tree tells `pc` is incremented twice.
```
    1:2:2:2:2: ASSIGN 0x564001ed1ff0 <e3184> {c9ao} @dt=0x564001ed3110@(sw32)
    1:2:2:2:2:1: ADD 0x564001ec9bf0 <e1335> {c9az} @dt=0x564001ed3110@(sw32)                                                                                 
    1:2:2:2:2:1:1: CONST 0x564001ed9560 <e1889> {c9bb} @dt=0x564001ecdb80@(G/sw32)  32'sh1
    1:2:2:2:2:1:2: VARREF 0x564001edb400 <e5993#> {c9aq} @dt=0x564001ed34e0@(w32)  t__DOT__state pkg=0x564001ec81d0 [RV] <- VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x564001edb010 <e5994#> {c9af} @dt=0x564001ed34e0@(w32)  t__DOT__state pkg=0x564001ec81d0 [LV] => VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2: ASSIGN 0x564001edb510 <e1349> {c10af} @dt=0x564001ed3110@(sw32)
    1:2:2:2:2:1: CONST 0x564001ecd510 <e1347> {c10am} @dt=0x564001ecdb80@(G/sw32)  32'sh2
    1:2:2:2:2:2: VARREF 0x564001edb120 <e1348> {c10af} @dt=0x564001ed3110@(sw32)  __Vfunc_B__1__Vfuncout [LV] => VARSCOPE 0x564001ec84f0 <e5984#> {c8ao} @dt=0x564001ed3110@(sw32)  TOP->__Vfunc_B__1__Vfuncout -> VAR 0x564001ec8b80 <e5981#> {c8ao} @dt=0x564001ed3110@(sw32)  __Vfunc_B__1__Vfuncout BLOCKTEMP
    1:2:2:2:2: ASSIGN 0x564001ed58f0 <e3172> {c4ao} @dt=0x564001ed3110@(sw32)
    1:2:2:2:2:1: ADD 0x564001ed5a90 <e1307> {c4az} @dt=0x564001ed3110@(sw32)                                                                                 
    1:2:2:2:2:1:1: CONST 0x564001ed1e30 <e1869> {c4bb} @dt=0x564001ecdb80@(G/sw32)  32'sh1
    1:2:2:2:2:1:2: VARREF 0x564001ecfe80 <e5962#> {c4aq} @dt=0x564001ed34e0@(w32)  t__DOT__state pkg=0x564001ec81d0 [RV] <- VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x564001ed52d0 <e5963#> {c4af} @dt=0x564001ed34e0@(w32)  t__DOT__state pkg=0x564001ec81d0 [LV] => VARSCOPE 0x564001ed0d20 <e4047> {c15aj} @dt=0x564001ed34e0@(w32)  TOP->t__DOT__state -> VAR 0x564001ecea30 <e620> {c15aj} @dt=0x564001ed34e0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2: ASSIGN 0x564001ed60c0 <e1321> {c5af} @dt=0x564001ed3110@(sw32)
    1:2:2:2:2:1: CONST 0x564001ed9750 <e1319> {c5am} @dt=0x564001ecdb80@(G/sw32)  32'sh1
    1:2:2:2:2:2: VARREF 0x564001ecfba0 <e1320> {c5af} @dt=0x564001ed3110@(sw32)  __Vfunc_A__0__Vfuncout [LV] => VARSCOPE 0x564001ecd250 <e5953#> {c3ao} @dt=0x564001ed3110@(sw32)  TOP->__Vfunc_A__0__Vfuncout -> VAR 0x564001ed0fe0 <e5950#> {c3ao} @dt=0x564001ed3110@(sw32)  __Vfunc_A__0__Vfuncout BLOCKTEMP
```

Adding `$display` prevents transformation from if-else to ternary operator and shows the correct result.

weisrc, 2022-04-23:
If I understand, the ternary operator and the if condition are not equivalent in terms of their mechanics. Ternary operates similar to a multiplexer, therefore executing whatever is inside and only taking its return value. If conditions behaves as expected. The issue is that Verilator is optimizing too much. It should not perform this transformation if what is within has side effects.

Correct me if I am wrong. 

Thank you for adding the tree dump to this issue.

yTakatsukasa, 2022-04-23:
SV standard explains Conditional operator as:
> If cond_predicate is true, the operator returns the value of the first expression without evaluating the second
expression; if false, it returns the value of the second expression without evaluating the first expression.

So I think converting if-else to AstCond is fine.

dougr33d, 2022-04-23:
The LRM (4.2.12 Conditional operator) appears to confirm that we should only be evaluating one side of the conditional, at least as of the 2014 Verilog LRM:

> The evaluation of a conditional operator begins with the evaluation of expression1. If expression1 evaluates to false (0), then expression3 is evaluated and used as the result of the conditional expression. If expression1 evaluates to true (any value other than zero (0)), then expression2 is evaluated and used as the result.

https://www.accellera.org/images/downloads/standards/v-ams/VAMS-LRM-2-4.pdf

wsnyder, 2022-04-24:
>It should not perform this transformation if what is within has side effects.

Correct. Verilator folliws what the HRM says, treating imcond like as an if, but a presumably broken optimization is violating that.

weisrc, 2022-04-24:
I tried looking for the file responsible for this optimization and couldn't find it. Can someone indicate me where to look? I tried looking in visitors with Branch and Cond in it.

yTakatsukasa, 2022-04-26:
Unfortunately there is no AstCond in V3Task.cpp

Because there is no good match for AstCond, the following line is called for it.
https://github.com/verilator/verilator/blob/2b91d764b55e3256bf0c2a5320cf18e86d4d831e/src/V3Task.cpp#L1538

`iterateChildren` simply visits nodes of `condition` and `expr1` and `expr2`.

`expr1` and `expr2` are both calling a function, the following `visit()` will be called for each.
https://github.com/verilator/verilator/blob/2b91d764b55e3256bf0c2a5320cf18e86d4d831e/src/V3Task.cpp#L1384
Then the visitor changes AST to evaluate both `expr1` and `expr2` even though they are not `pure`.

I think the following things are necessary to fix.
1. Fix AstFTaskRef::isPure() to consider sideeffect properly
2. Convert AstCond to if-else when both `expr1` and `expr2` are not pure

Considering pureness in the following codeblock works if the input HDL uses if-else, but does not help if the input HDL uses Conditional operator as in https://github.com/verilator/verilator/issues/3395#issue-1211763740. This is why I think `2.` is necessary.
https://github.com/verilator/verilator/blob/2b91d764b55e3256bf0c2a5320cf18e86d4d831e/src/V3Const.cpp#L2839-L2851

wsnyder, 2022-04-26:
>Convert AstCond to if-else when both expr1 and expr2 are not pure

Cond stays as a cond all the way through to C++ and should properly short-circuit so there should be no reason to convert it.

yTakatsukasa, 2022-04-27:
The following function in V3Task.cpp changes AST to call both functions regardless of pureness.
https://github.com/verilator/verilator/blob/2b91d764b55e3256bf0c2a5320cf18e86d4d831e/src/V3Task.cpp#L1384-L1437

Here is the result of V3Task.
AstCond exists after the pass, but it switches the result of inlined function. (if functions are not inlined, but still both are called).

```
    1:2:2:2:2: COMMENT 0x55fe4930bd60 <e5973#> {c17bc}  Function: A                                                                                                                                                                                                               
    1:2:2:2:2: COMMENT 0x55fe49305de0 <e6002#> {c19ar}  Function: B    
    1:2:2:2:2: ASSIGN 0x55fe49301230 <e3184> {c9ao} @dt=0x55fe493022f0@(sw32)
    1:2:2:2:2:1: ADD 0x55fe492f90b0 <e1335> {c9az} @dt=0x55fe493022f0@(sw32)     
    1:2:2:2:2:1:1: CONST 0x55fe492fdbd0 <e1889> {c9bb} @dt=0x55fe492fd6d0@(G/sw32)  32'sh1
    1:2:2:2:2:1:2: VARREF 0x55fe4930c260 <e5993#> {c9aq} @dt=0x55fe493026c0@(w32)  t__DOT__state pkg=0x55fe492f76c0 [RV] <- VARSCOPE 0x55fe492f79b0 <e4047> {c15aj} @dt=0x55fe493026c0@(w32)  TOP->t__DOT__state -> VAR 0x55fe492ff420 <e620> {c15aj} @dt=0x55fe493026c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x55fe4930be70 <e5994#> {c9af} @dt=0x55fe493026c0@(w32)  t__DOT__state pkg=0x55fe492f76c0 [LV] => VARSCOPE 0x55fe492f79b0 <e4047> {c15aj} @dt=0x55fe493026c0@(w32)  TOP->t__DOT__state -> VAR 0x55fe492ff420 <e620> {c15aj} @dt=0x55fe493026c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2: ASSIGN 0x55fe4930c370 <e1349> {c10af} @dt=0x55fe493022f0@(sw32)
    1:2:2:2:2:1: CONST 0x55fe492fdab0 <e1347> {c10am} @dt=0x55fe492fd6d0@(G/sw32)  32'sh2
    1:2:2:2:2:2: VARREF 0x55fe4930bf80 <e1348> {c10af} @dt=0x55fe493022f0@(sw32)  __Vfunc_B__1__Vfuncout [LV] => VARSCOPE 0x55fe493016d0 <e5984#> {c8ao} @dt=0x55fe493022f0@(sw32)  TOP->__Vfunc_B__1__Vfuncout -> VAR 0x55fe492f8040 <e5981#> {c8ao} @dt=0x55fe493022f0@(sw32)  __Vfunc_B__1__Vfuncout BLOCKTEMP
    1:2:2:2:2: ASSIGN 0x55fe49305790 <e3172> {c4ao} @dt=0x55fe493022f0@(sw32)
    1:2:2:2:2:1: ADD 0x55fe492f4b80 <e1307> {c4az} @dt=0x55fe493022f0@(sw32)     
    1:2:2:2:2:1:1: CONST 0x55fe492fde10 <e1869> {c4bb} @dt=0x55fe492fd6d0@(G/sw32)  32'sh1
    1:2:2:2:2:1:2: VARREF 0x55fe4930a000 <e5962#> {c4aq} @dt=0x55fe493026c0@(w32)  t__DOT__state pkg=0x55fe492f76c0 [RV] <- VARSCOPE 0x55fe492f79b0 <e4047> {c15aj} @dt=0x55fe493026c0@(w32)  TOP->t__DOT__state -> VAR 0x55fe492ff420 <e620> {c15aj} @dt=0x55fe493026c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x55fe492ff640 <e5963#> {c4af} @dt=0x55fe493026c0@(w32)  t__DOT__state pkg=0x55fe492f76c0 [LV] => VARSCOPE 0x55fe492f79b0 <e4047> {c15aj} @dt=0x55fe493026c0@(w32)  TOP->t__DOT__state -> VAR 0x55fe492ff420 <e620> {c15aj} @dt=0x55fe493026c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2: ASSIGN 0x55fe492ff580 <e1321> {c5af} @dt=0x55fe493022f0@(sw32)
    1:2:2:2:2:1: CONST 0x55fe492fdcf0 <e1319> {c5am} @dt=0x55fe492fd6d0@(G/sw32)  32'sh1
    1:2:2:2:2:2: VARREF 0x55fe492ff8b0 <e1320> {c5af} @dt=0x55fe493022f0@(sw32)  __Vfunc_A__0__Vfuncout [LV] => VARSCOPE 0x55fe49305620 <e5953#> {c3ao} @dt=0x55fe493022f0@(sw32)  TOP->__Vfunc_A__0__Vfuncout -> VAR 0x55fe49300220 <e5950#> {c3ao} @dt=0x55fe493022f0@(sw32)  __Vfunc_A__0__Vfuncout BLOCKTEMP
    1:2:2:2:2: ASSIGN 0x55fe492fcdf0 <e5972#> {c17ba} @dt=0x55fe493022f0@(sw32)
    1:2:2:2:2:1: COND 0x55fe493077e0 <e3199> {c17bc} @dt=0x55fe493022f0@(sw32)     
    1:2:2:2:2:1:1: VARREF 0x55fe49306ee0 <e3195> {c17aj} @dt=0x55fe49301920@(G/w1)  condition [RV] <- VARSCOPE 0x55fe49308e60 <e4033> {c13bn} @dt=0x55fe49301920@(G/w1)  TOP->condition -> VAR 0x55fe492fe5b0 <e1851> {c13bn} @dt=0x55fe49301920@(G/w1)  condition [PI] INPUT [P] [VSTATIC]  PORT
    1:2:2:2:2:1:2: VARREF 0x55fe4930c150 <e5970#> {c17bc} @dt=0x55fe493022f0@(sw32)  __Vfunc_A__0__Vfuncout [RV] <- VARSCOPE 0x55fe49305620 <e5953#> {c3ao} @dt=0x55fe493022f0@(sw32)  TOP->__Vfunc_A__0__Vfuncout -> VAR 0x55fe49300220 <e5950#> {c3ao} @dt=0x55fe493022f0@(sw32)  __Vfunc_A__0__Vfuncout BLOCKTEMP
    1:2:2:2:2:1:3: VARREF 0x55fe492f8c50 <e6001#> {c19ar} @dt=0x55fe493022f0@(sw32)  __Vfunc_B__1__Vfuncout [RV] <- VARSCOPE 0x55fe493016d0 <e5984#> {c8ao} @dt=0x55fe493022f0@(sw32)  TOP->__Vfunc_B__1__Vfuncout -> VAR 0x55fe492f8040 <e5981#> {c8ao} @dt=0x55fe493022f0@(sw32)  __Vfunc_B__1__Vfuncout BLOCKTEMP
    1:2:2:2:2:2: VARREF 0x55fe492ff750 <e622> {c17au} @dt=0x55fe493022f0@(sw32)  t__DOT__value [LV] => VARSCOPE 0x55fe49309050 <e4043> {c14af} @dt=0x55fe493022f0@(sw32)  TOP->t__DOT__value -> VAR 0x55fe492ff1a0 <e614> {c14af} @dt=0x55fe493022f0@(sw32)  t__DOT__value [VSTATIC]  VAR
```

I think if-else is necessary here to keep shortcut behavior.

gezalore, 2022-04-27:
Ah, so is this simply because we need to/chose to inline both function then just pick up the result with the ternary AstCond? If so then the AstIf is indeed warranted.

gezalore, 2022-04-27:
Note: probably same bug then applies to any short circuiting expression with impure function calls eg `a() && b()`

yTakatsukasa, 2022-04-27:
FYI here is the AST if functions are not inlined, but still they are called before AstCond.
```
    1:2:2:2:2: COMMENT 0x55f96f585d60 <e5985#> {c17bc}  Function: A         
    1:2:2:2:2: COMMENT 0x55f96f57fde0 <e6027#> {c19ar}  Function: B                       
    1:2:2:2:2: CCALL 0x55f96f5760f0 <e6018#> {c19ar} __VnoInFunc_B_TOP____024unit => CFUNC 0x55f96f583220 <e6006#> {c8ao}  __VnoInFunc_B_TOP____024unit
    1:2:2:2:2:2: VARREF 0x55f96f580a00 <e6019#> {c19at} @dt=0x55f96f57c6c0@(w32)  t__DOT__state [LV] => VARSCOPE 0x55f96f5719b0 <e4047> {c15aj} @dt=0x55f96f57c6c0@(w32)  TOP->t__DOT__state -> VAR 0x55f96f579420 <e620> {c15aj} @dt=0x55f96f57c6c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x55f96f584930 <e6022#> {c19ar} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_B__1__Vfuncout [LV] => VARSCOPE 0x55f96f576020 <e6015#> {c8ao} @dt=0x55f96f57c2f0@(sw32)  TOP->__Vfunc_B__1__Vfuncout -> VAR 0x55f96f575ec0 <e6012#> {c8ao} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_B__1__Vfuncout BLOCKTEMP
    1:2:2:2:2: CCALL 0x55f96f571e40 <e5974#> {c17bc} __VnoInFunc_A_TOP____024unit => CFUNC 0x55f96f577910 <e5962#> {c3ao}  __VnoInFunc_A_TOP____024unit
    1:2:2:2:2:2: VARREF 0x55f96f5811d0 <e5975#> {c17be} @dt=0x55f96f57c6c0@(w32)  t__DOT__state [LV] => VARSCOPE 0x55f96f5719b0 <e4047> {c15aj} @dt=0x55f96f57c6c0@(w32)  TOP->t__DOT__state -> VAR 0x55f96f579420 <e620> {c15aj} @dt=0x55f96f57c6c0@(w32)  t__DOT__state [VSTATIC]  VAR
    1:2:2:2:2:2: VARREF 0x55f96f586260 <e5978#> {c17bc} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_A__0__Vfuncout [LV] => VARSCOPE 0x55f96f57acc0 <e5971#> {c3ao} @dt=0x55f96f57c2f0@(sw32)  TOP->__Vfunc_A__0__Vfuncout -> VAR 0x55f96f572040 <e5968#> {c3ao} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_A__0__Vfuncout BLOCKTEMP  
    1:2:2:2:2: ASSIGN 0x55f96f576df0 <e5984#> {c17ba} @dt=0x55f96f57c2f0@(sw32)
    1:2:2:2:2:1: COND 0x55f96f5817e0 <e3199> {c17bc} @dt=0x55f96f57c2f0@(sw32)
    1:2:2:2:2:1:1: VARREF 0x55f96f580ee0 <e3195> {c17aj} @dt=0x55f96f57b920@(G/w1)  condition [RV] <- VARSCOPE 0x55f96f582e60 <e4033> {c13bn} @dt=0x55f96f57b920@(G/w1)  TOP->condition -> VAR 0x55f96f5785b0 <e1851> {c13bn} @dt=0x55f96f57b920@(G/w1)  condition [PI] INPUT [P] [VSTATIC]  PORT
    1:2:2:2:2:1:2: VARREF 0x55f96f585e70 <e5982#> {c17bc} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_A__0__Vfuncout [RV] <- VARSCOPE 0x55f96f57acc0 <e5971#> {c3ao} @dt=0x55f96f57c2f0@(sw32)  TOP->__Vfunc_A__0__Vfuncout -> VAR 0x55f96f572040 <e5968#> {c3ao} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_A__0__Vfuncout BLOCKTEMP
    1:2:2:2:2:1:3: VARREF 0x55f96f584a40 <e6026#> {c19ar} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_B__1__Vfuncout [RV] <- VARSCOPE 0x55f96f576020 <e6015#> {c8ao} @dt=0x55f96f57c2f0@(sw32)  TOP->__Vfunc_B__1__Vfuncout -> VAR 0x55f96f575ec0 <e6012#> {c8ao} @dt=0x55f96f57c2f0@(sw32)  __Vfunc_B__1__Vfuncout BLOCKTEMP
    1:2:2:2:2:2: VARREF 0x55f96f579750 <e622> {c17au} @dt=0x55f96f57c2f0@(sw32)  t__DOT__value [LV] => VARSCOPE 0x55f96f583050 <e4043> {c14af} @dt=0x55f96f57c2f0@(sw32)  TOP->t__DOT__value -> VAR 0x55f96f5791a0 <e614> {c14af} @dt=0x55f96f57c2f0@(sw32)  t__DOT__value [VSTATIC]  VAR
```

> Note: probably same bug then applies to any short circuiting expressions with impure function calls eg a() && b()

I'm afraid so.

It is possible to add a member variable to TaskVisitor to tell whether the visitor is in shortcut op context or not.
When in shortcut context, convert to if-else or keep the structure to the final C++ code.
The latter one sounds simpler if possible.

gezalore, 2022-04-27:
> When in shortcut context, convert to if-else or keep the structure to the final C++ code.

FWIW, for future optimizations I am planning to work on it would be good to not have short circuiting expressions, so I would be strongly for always converting these. Similarly, at some point, calls should only appear either as statement (void or return value ignored), or alone on the right hand side of assignments.

weisrc, 2022-04-27:
Agreed, the current generated code may be more efficient on a FPGA because it doesn’t need to use additional gates to enable/disable the logic. However in a computer, it’s unnecessary to execute code for a result that will be disposed right after.

gezalore, 2022-04-27:
Note: If converting all AstCond to AstIF V3MergeCond should be updated so it can still merge the then common
```
if (cond)
  a = x;
else
  a = y;
```

wsnyder, 2022-04-27:
I believe removing conditionals will be bad for performance, and suspect we will still need many other optimizations e.g. LogOr etc that will have to implement short-circuiting. Perhaps there's some meta concept proposed that will simplify that, but think that's an issue perhaps better described in a new/different thread as isn't something that is immediately needed related to fixing this.

wsnyder, 2022-05-14:
Thinking about the Cond-to-If another idea is to use AstExprStmt, so we can have the function statements "inline" without needing to convert away the Cond from the expression which is hard.

gezalore, 2022-05-15:
> I believe removing conditionals will be bad for performance

I think this is not true, in particular, clang lowers the conditional in the front-end, so the optimizer [never sees the conditional expression](https://godbolt.org/#g:!((g:!((g:!((h:codeEditor,i:(filename:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,selection:(endColumn:7,endLineNumber:5,positionColumn:7,positionLineNumber:5,selectionStartColumn:7,selectionStartLineNumber:5,startColumn:7,startLineNumber:5),source:'extern+int+r,+c,+t,+e%3B%0A%0Avoid+f()+%7B+r+%3D+c+%3F+t+:+e%3B+%7D%0A%0Avoid+g()+%7B+if+(c)+r+%3D+t%3B+else+r+%3D+e%3B+%7D'),l:'5',n:'0',o:'C%2B%2B+source+%231',t:'0')),k:26.875543951261967,l:'4',n:'0',o:'',s:0,t:'0'),(g:!((h:compiler,i:(compiler:clang1300,filters:(b:'0',binary:'1',commentOnly:'0',demangle:'0',directives:'0',execute:'1',intel:'0',libraryCode:'0',trim:'1'),flagsViewOpen:'1',fontScale:14,fontUsePx:'0',j:1,lang:c%2B%2B,libs:!(),options:'-emit-llvm+-O0+-mllvm+--print-before-all',selection:(endColumn:1,endLineNumber:1,positionColumn:1,positionLineNumber:1,selectionStartColumn:1,selectionStartLineNumber:1,startColumn:1,startLineNumber:1),source:1,tree:'1'),l:'5',n:'0',o:'x86-64+clang+13.0.0+(C%2B%2B,+Editor+%231,+Compiler+%231)',t:'0')),k:39.791122715404704,l:'4',m:100,n:'0',o:'',s:0,t:'0'),(g:!((h:output,i:(compilerName:'x86-64+clang+13.0.0',editorid:1,fontScale:14,fontUsePx:'0',j:1,wrap:'0'),l:'5',n:'0',o:'Output+of+x86-64+clang+13.0.0+(Compiler+%231)',t:'0')),k:33.33333333333333,l:'4',n:'0',o:'',s:0,t:'0')),l:'2',n:'0',o:'',t:'0')),version:4)