verilator/verilator #3397: Support for module level property statements

supleed2, 2022-04-25:
I was wondering if it was possible (and the level of effort needed) to support module level property statements, rather than just inline properties within an `assert property ()` block. I did some initial searching and didn't find any other issue, but I believe this is the relevant area within the parser: https://github.com/verilator/verilator/blame/master/src/verilog.y#L5248

I'm not sure whether the logic needed for the commented out section is already there as I am not familiar with the structure of the project yet. I have a minimal testcase and the resulting error here:
```sv
module testcase (
  input var clk,
  input var rst_n
);
  logic a;
  always_ff @(posedge clk or negedge rst_n)
    if (!rst_n) a <= 0;
    else a = ~a;
  property pr_001;
    @(posedge clk) !rst_n |-> !a;
  endproperty
endmodule
```
```bash
$ verilator --lint-only testcase.sv 
%Error: testcase.sv:9:3: syntax error, unexpected property
    9 |   property pr_001;
      |   ^~~~~~~~
%Error: Cannot continue
```

wsnyder, 2022-04-26:
Note what you have is only declaring it, presumably you would also be using it somewhere for it to have a purpose.

my_assert : assert property (pr_001);

There's no current support for named properties or using them.  I don't expect there's much complication in doing this, but there would be probably a week of plumbing to parse them (mostly there in comments), make Ast nodes of them, then resolve them, etc.

wsnyder, 2022-04-26:
P.S. if you are just trying to make an assertion at module scope that works, the syntax is this:

       assert property @(posedge clk) !rst_n |-> !a;

supleed2, 2022-04-26:
Ah yes, I missed a line, this should be better:
```sv
module t (
  input  var i_clk,
  input  var i_rst_n,
  output var o_a
);
  always_ff @(posedge i_clk or negedge i_rst_n)
    if (!i_rst_n) o_a <= 0;
    else o_a = ~o_a;
  property pr_001;
    @(posedge i_clk) !i_rst_n |-> !o_a;
  endproperty
  test_assert : assert property (pr_001) else $error("pr_001 failed");
endmodule
```
Thanks for the extra note too,  I was trying something like
```sv
assert property (@(posedge clk) !rst_n |-> !a) else begin
  $info("More info");
  $error("Error name");
end
```
earlier, which I assume is just incorrect syntax then?

Is there anything else I should provide?

wsnyder, 2022-04-27:
The syntax you tried is technically legal, but at present Verilator is very limited in what it can support so the parser likewise matches what's supported instead of what is legal.

varunr71, 2023-02-06:
Hi All,
Looks like this issue is from 2022. Has there been any more updates on the support for SVA in verilator?

wsnyder, 2023-02-06:
Property statements were supported with #3667 in 5.004, forgot to close this.

In general we don't well document what's missing, best route is to try the code and file a bug if there isn't already one on the missing feature of interest.

wsnyder, 2023-02-06:
Test case at top tested and works now.

closed 2023-02-06 by wsnyder