verilator/verilator #3399: Incorrect tristate enable logic

shareefj, 2022-04-27:
Version:  Verilator 4.214 2021-10-17 rev v4.214
OS: RHEL

Hi,

I've got a bug or user issue, still to be determined, but would like some help in debugging further.  I have a simple tristate model that I'm using for both FPGA and Verilator and I'm seeing some strange behaviour in Verilator where the output port of the tristate, the input to the core, isn't being driven when enabled.  I tried to construct a testcase using the tristate model on it's own but that works fine so it seems to be some interaction with the larger system.

My tristate model looks like:

```systemverilog
module pad_tri (
  output logic pad_o,
  input logic pad_i,
  input logic oe,
  input logic ie,
  inout wire pad
);

  assign pad   = oe ? pad_i : 'z;
  assign pad_o = ie ? pad   : 'z;

endmodule
```

What I can see in the waves is that `pad` is busy toggling while `ie` is high but I get no output on `pad_o`.

I modified my toplevel to just instantiate a single `pad_tri` rather than have them inside of generate statements to see if that helped but no change.  I've been looking through the generated CPP files and the following looks strange.  I've not looked at the naming conventions here before but the following looks dodgy to me:

```cpp
    vlSelf->chip_sim_tb__DOT__u_led_2_tristate__DOT__pad_o__out__en0 
        = (1U & (((IData)(vlSelf->chip_sim_tb__DOT__led_input_en) 
                  & (IData)(vlSelf->chip_sim_tb__DOT__led_output_en)) 
                 >> 2U));
```
The nets `led_input_en` and `led_output_en` are connected to `ie` and `oe` respectively.

Having seen that, I've just removed modified the `pad_o` assignment to be just `assign pad_o = pad` and the simulation works correctly.  So could this just be a broken optimisation?

Can you also explain what the various signal suffixes mean. Is `__out__en0` meant to be the output enable?

wsnyder, 2022-04-27:
I agree this smells like a broken optimization or tristate error, we'll need a test case to debug it further.

__en0 means this is the enable for the 0th (first) expression that can drive into the given resolved tristate net.  If there's multiple drivers those will have multiple __en terms.  But you're seeing the result after many optimizations, it's better to run with --debug and see what's in the .tree files in the expression which is before optimization.

HungMingWu, 2022-04-28:
We met the similar issue previously,  I made a workable version.
https://github.com/HungMingWu/verilator/tree/tristate
Check whether it works.

shareefj, 2022-04-29:
Thanks @HungMingWu 

I just gave your version a go and after commenting out some code it didn't like, I get the same result.  Thanks anyway.

@wsnyder Not sure if I'll be able to create a testcase or not.  I guess the one I tried to create didn't do anything so perhaps I'll try adding a real tristate driver on the other end.

shareefj, 2022-04-29:
@wsnyder Looks like I managed to get one:  https://bitbucket.org/shareefjalloqidex/verilator_bugs/src/verilator_3399/

```bash
cd cloned_repo && make
grep -A 1 pad_o__out__en0 build/verilator_test_case_0/sim-verilator/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp | head -n 2
```

wsnyder, 2022-05-14:
Can you please make this into an example that has no dependencies besides Verilator, thanks.

shareefj, 2022-05-16:
Sorry, done.  `make` to Verilate, run and launch GtkWave, or just `make build` to run Verilator.

shareefj, 2022-05-16:
As an aside, I've just been clearing up my local area to merge a PR and found that the model suddenly stopped working.  While I was debugging the above issue, I wrote a C logger to access the internal state of the SW-DP module.  To access these signals I had to make them public.

Having just remove all my local hacks, including the public signals, the model stops working.  So there may be more than one issue here.  I'll see if I get time before Friday to look at this and compare the generate code but then I'm out on holiday for a week so sorry for the radio silence if so.

wsnyder, 2022-05-16:
Thanks for refactoring. Making everything public changes a lot of ordering and other effects, so that might be unrelated.

vighneshiyer, 2022-07-28:
I think we've run into a similar bug when trying to bump the Verilator version in Chipyard (https://github.com/ucb-bar/chipyard/pull/1205). It appears that a tristate expression like this one:
```
    assign ts_0 = (drive[0] ? d[0] : 1'bz);
    assign ts_1 = (drive[1] ? d[1] : 1'bz);
```
always drives `ts_1` to 0, even if `drive[1] == 1 && d[1] == 1`. `ts_0` behaves as expected. This behavior only occurs when the `drive` and `d` buses are more than 1 bit wide, so it seems to be an issue with multibit optimization.

I have reproduced the bug in an isolated and reduced test case here (reproduction steps and logs in the readme): https://github.com/vighneshiyer/verilator-tristate-bug

I bisected and found that the bug was introduced in commit 8681861b.  @gezalore Do you know what the issue could be?

gezalore, 2022-07-28:
There have been a couple of fixes to that that went in recently. If the issue still happens on mater, then we will need to take a look. Maybe @yTakatsukasa would be willing to help?

yTakatsukasa, 2022-07-28:
OK. I will take a look at the testcase from @vighneshiyer this weekend.

vighneshiyer, 2022-07-28:
Thanks for the patch @yTakatsukasa! I confirmed that it fixes the small testcase and the larger Chipyard test that was failing on Verilator master.

yTakatsukasa, 2022-07-28:
Thank for testing.
I suspect there are similar issues that triggers bit op tree optimization in inappropriate places.

yTakatsukasa, 2022-07-30:
What is happening in https://github.com/verilator/verilator/issues/3399#issuecomment-1113567274 is not optimization related, but by tri-state handling.
I'm not so familiar with V3Tristate, but I made a patch that passes  https://github.com/verilator/verilator/issues/3399#issuecomment-1113567274 and all t_regress tests.
I will send a PR after making tests.

commit d20f22be (2022-08-07): Fix tristate logic when reading inout port in a module #3399 (#3523)

* Tests: Add a test to reproduce #3399

* Fix #3399. When reading an inout port in a module, it should refer the
original inout port, not the generated MODTEMP.
closed 2022-08-07 by yTakatsukasa