verilator/verilator #3401: How can i link verilator(system default -std=c++14) to libsystemc(-std=c++11)

muyuer0, 2022-04-28:
i tries to use -CFLAGS -LDFLAGS but not work，does it mean once i installed verilator， it only support specified c++ stardard?

err info:
```
g++ -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11   -I/usr/local/systemc-2.3.3/include -std=gnu++14 -Os -c -o Vour__ALL.o Vour__ALL.cpp

Vour__ALL.cpp:(.text.startup+0x35): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201402L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201402L(sc_core::sc_writer_policy)'
```

wsnyder, 2022-04-28:
This is unrelated to Verilator itself and more a C++/glibc question.  I think it's complicated to ever mix glibc versions in one program and I have not done it, but I found this which suggests it is possible. https://developers.redhat.com/blog/2019/08/01/how-the-gnu-c-library-handles-backward-compatibility

wsnyder, 2022-04-28:
BTW if you don't think you are **intending** to mix versions, then probably your cflags or ldflags are not picking up the right options.

muyuer0, 2022-04-29:
thanks a lot. maybe i misunderstood sth，
words below in verilator_doc chapter verilating->cmake part， make me think i can use systemc-version to compile verilate-systemc and systemc code? If its right, how? For example? 
```
verilator -Wall --sc our.v
make -j -f Vour.mk Vour__ALL.a
```
```
SYSTEMC
Optional. Enables SystemC mode, defaults to C++ if not specified.
When using Accellera’s SystemC with CMake support, a CMake target is available that simplifies the SystemC
steps. This will only work if the SystemC installation can be found by CMake. This can be configured by setting
the CMAKE_PREFIX_PATH variable during CMake configuration.
Don’t forget to set the same C++ standard for the Verilated sources as the SystemC library. This can be specified
using the SYSTEMC_CXX_FLAGS environment variable.
```

wsnyder, 2022-05-01:
Are you using Ubuntu 22.04 LTS?  It looks like this might be a mistake they made building it - the proper dependencies etc are not in the libsystemc package and/or they built using a different compiler than the default now is.  Perhaps you could track down the appropriate way to report a bug on them?

FanShupei, 2022-05-03:
libstdc++/glibc tries hard to maintain abi compatibility among different C++ standard. For example, if some parts of your program are compiled with `-std=c++11`, while other parts are compiled with `-std=c++14`, it should just work by linking them together (provided that libstdc++ in your system supports at least c++14).

However, considering the complexity of c++ abi, systemc intentionally forbids mixing up c++ standard. In your case, libsystemc is NOT compiled with `-std=c++14`, but your program is compiled with `-std=c++14`, so you get the linking error. The only thing you should do is compiling your program with `-std=c++11`.

Technically, Verilator generates c++ source code, you could always compile it with proper compile options. However, there's probability that the Makefile generated by Verilator messes something up , so the flow doesn't work out of box.

I could dig into it deeper, to analysis whether the misconfiguration comes from Verilator side. But definitely need you to provide more information.
Your distro version: ubuntu 18.04/20.04/22.04? or others
Your systemc version: install by `apt install` ? or compiled by yourself?
Your verilator version: install by `apt install`? or compiled by yourself?
Does your systemc work without Verilator. If you write a simple systemc program, it just works? or observes similar linking error?
How do you compile your verilated model: by just invoking Makefile generated by Verilator? or you compile it manually?

muyuer0, 2022-05-11:
thanks！
Your distro version: ubuntu 18.04/20.04/22.04? or others

> 18.04

Your systemc version: install by apt install ? or compiled by yourself?

> by myself，std=c++11

Your verilator version: install by apt install? or compiled by yourself?

> by myself, std = gnu++14

Does your systemc work without Verilator. If you write a simple systemc program, it just works? or observes similar linking error?

> works well itself. If i compiler verilator in c++11, systemcs also works well with verilator.

How do you compile your verilated model: by just invoking Makefile generated by Verilator? or you compile it manually?

> use makefile generated by verilator，only one 'our.v' code，one 'sc_main.cpp' code, compiler as
`verilator -Wall --sc our.v
make -j -f Vour.mk Vour__ALL.a //Vour.mk generated by verilator
make -j -f Vour.mk ../sc_main.o verilated.o
g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o -o Vour -lsystemc 
// (tried to use CFLAGS/LDFLAGS to specify c++ verison, but not works)`

muyuer0, 2022-05-11:
verilator -Wall --sc our.v 
make -j -f Vour.mk Vour__ALL.a //Vour.mk generated by verilator 
make -j -f Vour.mk ../sc_main.o verilated.o 
g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o -o Vour -lsystemc
 // (tried to use CFLAGS/LDFLAGS to specify c++ verison, but not works)

wsnyder, 2022-05-12:
Given Verilated code needs to use the SystemC headers I think you must compile systemc and verilated code with the same headers.  So you'll need to either change what you use for your systemc, or what you use for verilated code.

closed 2022-05-12 by wsnyder