verilator/verilator #3407: Memory leak at V3FileLine

HungMingWu, 2022-05-07:
Here is the message on valgrind
``` text
==11991== 64 bytes in 1 blocks are definitely lost in loss record 68 of 227
==11991==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==11991==    by 0x45403A: FileLine::copyOrSameFileLine() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69A532: V3LexerBase::yylex() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641194: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x46CE4E: V3Global::readFiles() (in /home/hm/verilator/bin/verilator_bin)
==11991==
==11991== 64 bytes in 1 blocks are definitely lost in loss record 69 of 227
==11991==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==11991==    by 0x45403A: FileLine::copyOrSameFileLine() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x692FDF: V3LexerBase::yylex() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640C47: V3ParseImp::tokenPeekp(unsigned long) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640F59: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==
==11991== 64 bytes in 1 blocks are definitely lost in loss record 70 of 227
==11991==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==11991==    by 0x45403A: FileLine::copyOrSameFileLine() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69A6B8: V3LexerBase::yylex() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640C47: V3ParseImp::tokenPeekp(unsigned long) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640F59: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==
==11991== 64 bytes in 1 blocks are definitely lost in loss record 71 of 227
==11991==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==11991==    by 0x45403A: FileLine::copyOrSameFileLine() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69A6B8: V3LexerBase::yylex() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641194: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x46CE4E: V3Global::readFiles() (in /home/hm/verilator/bin/verilator_bin)
==11991==
==11991== 64 bytes in 1 blocks are definitely lost in loss record 72 of 227
==11991==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==11991==    by 0x45403A: FileLine::copyOrSameFileLine() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69A2E9: V3LexerBase::yylex() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641194: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
[... 118 lines cut ...]
==11991==    by 0x69E592: V3ParseImp::yylexReadTok() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640983: V3ParseImp::tokenPull() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640C47: V3ParseImp::tokenPeekp(unsigned long) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x640F59: V3ParseImp::tokenPipeline() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x641CFC: V3ParseImp::tokenPipelineSym() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6424C3: V3ParseImp::tokenToBison() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x64A60B: yyparse() (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
[... cut ...]
==11991==    by 0x6407B1: V3ParseImp::lexFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
==11991==    by 0x6427A6: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/hm/verilator/bin/verilator_bin)
```
Most comes from `FileLine::copyOrSameFileLine`, it create a new copy, but no one destroy it.

HungMingWu, 2022-05-08:
Update, here is the result of enable `VL_LEAK_CHECKS=1`
``` text
$ valgrind --leak-check=full  /home/hm/verilator/bin/verilator_bin -cc top.v --debug-leak
==5211== Memcheck, a memory error detector
==5211== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==5211== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==5211== Command: /home/hm/verilator/bin/verilator_bin_dbg -cc top.v --debug-leak
==5211==
==5211==
==5211== HEAP SUMMARY:
==5211==     in use at exit: 136,492 bytes in 260 blocks
==5211==   total heap usage: 4,718 allocs, 4,458 frees, 567,117 bytes allocated
==5211==
==5211== 1,330 (112 direct, 1,218 indirect) bytes in 1 blocks are definitely lost in loss record 179 of 185
==5211==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5211==    by 0x583C9C: FileLine::newContent() (V3FileLine.cpp:158)
==5211==    by 0x9B61AA: V3PreProcImp::openFile(FileLine*, VInFilter*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3PreProc.cpp:806)
==5211==    by 0x794D7E: V3PreShellImp::preprocOpen(FileLine*, VInFilter*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3PreShell.cpp:148)
==5211==    by 0x796481: V3PreShellImp::preproc(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, VInFilter*, V3ParseImp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3PreShell.cpp:98)
==5211==    by 0x79454E: V3PreShell::preproc(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, VInFilter*, V3ParseImp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3PreShell.cpp:168)
==5211==    by 0x933D57: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3ParseImp.cpp:281)
==5211==    by 0x9349F2: V3Parse::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3ParseImp.cpp:592)
==5211==    by 0x5AF919: V3Global::readFiles() (V3Global.cpp:58)
==5211==    by 0x30E20C: verilate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (Verilator.cpp:603)
==5211==    by 0x30EEFC: main (Verilator.cpp:733)
==5211==
==5211== 1,457 (112 direct, 1,345 indirect) bytes in 1 blocks are definitely lost in loss record 181 of 185
==5211==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5211==    by 0x583C9C: FileLine::newContent() (V3FileLine.cpp:158)
==5211==    by 0x933D36: V3ParseImp::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3ParseImp.cpp:276)
==5211==    by 0x9349F2: V3Parse::parseFile(FileLine*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (V3ParseImp.cpp:592)
==5211==    by 0x5AF919: V3Global::readFiles() (V3Global.cpp:58)
==5211==    by 0x30E20C: verilate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (Verilator.cpp:603)
==5211==    by 0x30EEFC: main (Verilator.cpp:733)
==5211==
==5211== LEAK SUMMARY:
==5211==    definitely lost: 224 bytes in 2 blocks
==5211==    indirectly lost: 2,563 bytes in 24 blocks
==5211==      possibly lost: 0 bytes in 0 blocks
==5211==    still reachable: 133,705 bytes in 234 blocks
==5211==         suppressed: 0 bytes in 0 blocks
==5211== Reachable blocks (those to which a pointer was found) are not shown.
==5211== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==5211==
==5211== For lists of detected and suppressed errors, rerun with: -s
==5211== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```
And there is another  compile error issue on enable `VL_LEAK_CHECKS`, I will open another issue.

gezalore, 2022-05-14:
@wsnyder Why do we leak by default on the non-debug build (verilator_bin)? (I understand `--debug-leak`, but I thought that's for verilator_bin_dbg)

wsnyder, 2022-05-14:
Slowing runtime to avoid a leak for a structure only reclaimable at exit doesn't seem worth it, unless trying to find leaks of course.

gezalore, 2022-05-14:
I think that's premature optimization TBH, you would likely find freeing everything properly at the end is insignificant compared to running the actual compiler.

wsnyder, 2022-05-14:
You are assuming it's premature ;). Filelines are hot code with 10M++ constructions (or were hot as other things have gotten slower), which is how it got this way.

closed 2022-05-14 by wsnyder