verilator/verilator #3409: complex assign in always_comb

RaynardQiao, 2022-05-07:
version ： git master  2022/5/7
demo：
```
module twocounter( din,row_idx, row_found);
  input din;
  output [1:0]row_idx;
  output row_found; 
  reg [3:0] din;
  reg [3:0] wide_din;
  reg row_found;
  reg [1:0]row_idx;

always_comb
begin : pricen_2d
	integer x;
	wide_din = {4{1'b0}};
	wide_din[3:0] = din;

	row_idx = {2{1'b0}};
	row_found = 1'b0;
	for (x = 0; $unsigned(x) < 4; x=x+1)
	begin
		row_idx = !row_found?x[1:0] : row_idx;
		row_found = !row_found?wide_din[x]:row_found;
	end
end
endmodule
```

trans code:
```

    // Init
    CData/*0:0*/ __Vcellout__twocounter__row_found;
    CData/*1:0*/ __Vcellout__twocounter__row_idx;
    CData/*3:0*/ __Vcellinp__twocounter__din;
    CData/*3:0*/ twocounter__DOT__wide_din;
    // Body
    VL_ASSIGN_ISI(4,__Vcellinp__twocounter__din, vlSelf->din);
    twocounter__DOT__wide_din = __Vcellinp__twocounter__din;
    __Vcellout__twocounter__row_found = (1U & (IData)(twocounter__DOT__wide_din));
    __Vcellout__twocounter__row_idx = ((IData)(__Vcellout__twocounter__row_found)
                                        ? 0U : 1U);
    if (__Vcellout__twocounter__row_found) {
        __Vcellout__twocounter__row_found = (1U & (IData)(__Vcellout__twocounter__row_found));
        __Vcellout__twocounter__row_idx = __Vcellout__twocounter__row_idx;
    } else {
        __Vcellout__twocounter__row_found = (1U & ((IData)(twocounter__DOT__wide_din) 
                                                   >> 1U));
        __Vcellout__twocounter__row_idx = 2U;         //   Example  din 4'b0010, row idx should be 1, not 2U
    }
    if (__Vcellout__twocounter__row_found) {
        __Vcellout__twocounter__row_found = (1U & (IData)(__Vcellout__twocounter__row_found));
        __Vcellout__twocounter__row_idx = __Vcellout__twocounter__row_idx;
    } else {
        __Vcellout__twocounter__row_found = (1U & ((IData)(twocounter__DOT__wide_din) 
                                                   >> 2U));
        __Vcellout__twocounter__row_idx = 3U;
    }
    if (__Vcellout__twocounter__row_found) {
        __Vcellout__twocounter__row_found = (1U & (IData)(__Vcellout__twocounter__row_found));
        VL_ASSIGN_SII(1,vlSelf->row_found, 1U);
    } else {
        __Vcellout__twocounter__row_found = (1U & ((IData)(twocounter__DOT__wide_din) 
                                                   >> 3U));
        VL_ASSIGN_SII(1,vlSelf->row_found, 0U);
    }
    VL_ASSIGN_SII(2,vlSelf->row_idx, __Vcellout__twocounter__row_idx);
```
example:  For input 4'b0010(din), i think the output should be  row_id=1, 
But in the logic above, what comes out is 2U.

I have read the source code and preliminarily determine that the problem is in V3Life.cpp. However, I have no idea how to solve the problem ，need help

RaynardQiao, 2022-05-12:
The error is found in the V3MergeCond.cpp（visit AstNodeAssign） file and is being located continuously.

wsnyder, 2022-05-15:
@gezalore can you please look at this?
I made a test_regress test, attached

[t_0.v.txt](https://github.com/verilator/verilator/files/8694093/t_0.v.txt)

The problem goes away if I do this, but it looks like your intent was to handle these...

     -        return reason == Mergeable::NO_COND_ASSIGN;
     +//        return reason == Mergeable::NO_COND_ASSIGN;
     +        return false;
 .

gezalore, 2022-05-17:
Now fixed in master. Thanks to both for tracking this down and providing the test case.

commit 1a056f6d (2022-05-17): Fix invalid conditional merging when starting at 'c = c ? a : b'

Fixes #3409.
closed 2022-05-17 by gezalore