verilator/verilator #3414: Assign statement line coverage missing

phelter, 2022-05-12:
Environment: 
* verilator version - `Verilator 4.222 2022-05-02 rev v4.222-24-g4618fd5a`
* OS - Windows 11, wsl2 Ubuntu-18.04
* 
I have a simple circuit:

```verilog
module cell_inv (
    input  wire a,         //% a input.
    output wire z          //% z output, z = ~a;
    );
    assign z = ~a;
endmodule //cell_inv
```
I have a testbench that tests the karnaugh map and verilate it with `--coverage` to ensure both line and toggle coverage.

My C++ testbench instantiates the necessary verilator code and verilated cell_inv
```cpp
    std::unique_ptr<VerilatedContext> m_context; ///< Verilator Context
    std::unique_ptr<VerilatedVcdC>    m_trace;   ///< Trace Class variable
    std::unique_ptr<T>                m_dut;     ///< Device Under Test (DUT)
 ```
And these are setup appropriately for execution for vcd generation and writing out the coverage at the end of the test.

```cpp
m_dut->a = 0
m_context->timeInc(1);
m_dut->eval();
m_dut->a = 1
m_context->timeInc(1);
m_dut->eval()

//on completion:
m_context->coveragep()->write(m_covName.c_str());
```
I then use this to generate lcov/genhtml coverage and notice a few things:

1. there is no line coverage for the assign statement - is this expected?
The `verilog_coverage --annotate-all --annotate annotate test.data` command is called and the output is:

```
        module cell_inv (
%000002     input  wire a,         //% a input.
%000003     output wire z          //% z output, z = ~a;
            );
            assign z = ~a;
        endmodule //cell_buf
```

I see the coverage data file has only toggle instances, and there is no line coverage for the assignment.

Thanks.

A few bonus questions/suggestions:

1. When generating the genhtml - the line and toggle coverage is amalgamated into one group called `Hit`  I was expecting them to be separate groups - Line and toggle - which would be helpful.  eg: Line, Toggle, branch(later).
2. Also toggle coverage typically separates out `0->1` and `1->0` counts - but it's not a big deal having the aggregate like it is right now.

phelter, 2022-05-12:
A side note - can add this to another issue if you like:

Appreciate all the work on this.  I've integrated it into my environment and have a few suggestions for cleaning up the CMake integration for using verilated code, and improving the quality of code that is consumed by the testbenches/main files.
* Create a single Verilator::base static target that gets linked to instead of including the files individually to each build - this may require the `#defines` to become template parameters where used.
* Identified all of the `Wall, Wpedantic, Weffc++, Weverything` and clang-tidy issues in the code stored in there for compilers GCC 7.5 and Clang-13 (those are the versions I used) - there are a few gotcha's that might need cleanup.  First steps.  Next steps is to clean each of them up in the code base - Interested?

phelter, 2022-05-12:
I may be answering my own question - found a good list of coverage types for hdl language verification here:
https://www.theartofverification.com/coverage-2/

the Expression/conditional coverage is the one that's missing from the line.  It may be that expression coverage is not supported?

wsnyder, 2022-05-12:
>  there is no line coverage for the assign statement - is this expected?

assign statements are effectively executed every cycle by Verilator so there would be no purpose in line coverage on them.

>  When generating the genhtml - the line and toggle coverage is amalgamated into one group called Hit I was expecting them to be separate groups - Line and toggle - which would be helpful. eg: Line, Toggle, branch(later).

I don't believe genhtml can support those subtypes (as opposed to verilator coverage text output).  If there is a way, we'd welcome a patch to improve this.

>    Also toggle coverage typically separates out 0->1 and 1->0 counts - but it's not a big deal having the aggregate like it is right now.

Combining them results in half the coverage data and runtime impact, so this is what's currently done.

>the Expression/conditional coverage is the one that's missing from the line. It may be that expression coverage is not supported?

There is currently no expression coverage nor FSM coverage support.  We'd welcome contributions in these areas (note it's significant effort.)

Closing as answered, but feel free to follow up.

closed 2022-05-12 by wsnyder