verilator/verilator #3415: Arm tarmac instruction trace causes Verilator build to hang

shareefj, 2022-05-14:
This may well fall under the, "but it's not synthesisable", banner but I wanted to check as I'd expect an error rather than a hang.  I'm Verilating a small Cortex-M MCU that consumes 0.9GB of RAM and takes 54 seconds to Verilate, so a tiddler.  As soon as I add in the Arm tarmac module the RAM usage balloons to 10+GB and the build seems to hang indefinitely.

If you have access to any of the Arm M class CPUs then you've probably seen the tarmac logger before.  It's a simple top level Verilog module that monitors some hierarchical nets within the CPU and then prints out an instruction stream.  Aside from calling `$fopen` and `$fwrite`, the only other functions in there are string processing ones that concatenate the final string to print.

The functions are all of the following type:

```verilog
function [`STR_LEN:0] disass_func_1;
  input [`STR_LEN:0] in;
  begin
    o = in;

    // do some decoding of something
    o = { o, result_of_decode };

    // perhaps call a secondary function
    o = disass_func_2(o);

    // assign output
    disass_func_1 = o;

  end
endfunction
```

While typing this issue, I've left the compile running and the RAM usage remained stable at around 11GB but is now increasing to the limit of my machine which is 25GB.  The system killed it after 13m.

From my limited understanding of how the internals of Verilator work, could it be sat there trying to optimise a long list of string concatenations and falling over?  Just wanted to get some input before I put into any effort converting the 493 string functions to C.  Ergh.

wsnyder, 2022-05-14:
Those should work. It likely is either a bug where it never finishes and is recursing forever, or it will eventually finish but slowly due to some O(n^2) mistake.   You might want to try marking the function no inline.

Can you debug?  Try using --debug --gdb then wait, ctrl c, and "bt" to see where it is.

shareefj, 2022-05-14:
Thanks, will try that.  Might also try it outside of our system with a simple dummy CPU to simplify things.

wsnyder, 2022-05-14:
Note if you use "-E" you can get the preprocessor output, then feed that back in. That sometimes is easier to manipulate for experiments than huge directory trees with complicated search paths.

shareefj, 2022-05-14:
Seems to still be ticking over on the command line just before it runs out of memory.  This is the last few lines:

```
- V3Order.cpp:2047:     Domains...
dot -Tpdf -o ~/a.pdf ./Vchip_sim_tb_064_orderg_domain.dot
- V3Order.cpp:2066:     Set up mtasks...
- V3Partition.cpp:627:Writing ./Vchip_sim_tb_066_cp.txt
- V3Order.cpp:2071:     Sensitive...
dot -Tpdf -o ~/a.pdf ./Vchip_sim_tb_067_orderg_done.dot
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_068_order.tree
- V3GenClk.cpp:221:   genClkAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_069_genclk.tree
- V3Clock.cpp:446:    clockAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_070_clock.tree
- V3Const.cpp:3594:   constifyAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_071_const.tree
- V3Life.cpp:488:     lifeAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_072_life.tree
- V3LifePost.cpp:350: lifepostAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_073_life_post.tree
- V3Const.cpp:3594:   constifyAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_074_const.tree
- V3Dead.cpp:480:     deadifyAllScoped:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_075_deadAllScoped.tree
- V3Changed.cpp:241:  changedAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_076_changed.tree
- V3Trace.cpp:901:    traceAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_077_trace.tree
- V3Localize.cpp:221: localizeAll:
- V3Descope.cpp:283:  descopeAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_079_descope.tree
- V3Combine.cpp:233:  combineAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_080_combine.tree
- V3Const.cpp:3594:   constifyAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_081_const.tree
- V3Dead.cpp:474:     deadifyAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_082_deadAll.tree
- V3Clean.cpp:316:    cleanAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_083_clean.tree
- V3Premit.cpp:397:   premitAll:
- V3Ast.cpp:1108:     Dumping ./Vchip_sim_tb_084_premit.tree
- V3Expand.cpp:897:   expandAll:
^C
Program received signal SIGINT, Interrupt.
0x0000555555e23efc in tcmalloc::CentralFreeList::Populate() ()
#0  0x0000555555e23efc in tcmalloc::CentralFreeList::Populate() ()
#1  0x0000555555e24068 in tcmalloc::CentralFreeList::FetchFromOneSpansSafe(int, void**, void**) ()
#2  0x0000555555e24110 in tcmalloc::CentralFreeList::RemoveRange(void**, void**, int) ()
#3  0x0000555555e2b766 in tcmalloc::ThreadCache::FetchFromCentralCache(unsigned int, int, void* (*)(unsigned long)) ()
#4  0x000055555579d965 in AstTypeTable::findLogicBitDType (this=0x555556bce000, fl=0x5555624d6380, kwd=..., width=32, widthMin=32, numeric=...) at ../V3AstNodes.cpp:984
Backtrace stopped: Cannot access memory at address 0x7fffffffb6c8
```

wsnyder, 2022-05-14:
Does it pass with "verilator -Ox"?

Try "--debug --debugi-V3Expand 9" and pipe to a file.  Be ready to ctrl-C if your disk starts to fill.  See what the bottom of the file says it is doing.

shareefj, 2022-05-16:
Hi, I haven't tried your last suggestion but did some investigation yesterday and think I know the issue.  This SV code and all the functions are performing a decode of the instruction stream so at the top level there are lots of casez statements of the form;

```
    casez({&opcode[31:28], opcode[27:0]})

      29'b0_0000_000?_????_????_????_1001_???? : o = ual_a32_mul(opcode);
      29'b0_0000_1???_????_????_????_1001_???? : o = ual_a32_mull_mlal(opcode);
      29'b0_0000_0100_????_????_????_1001_???? : o = ual_a32_umaal(opcode);
      etc etc
```

I extracted one of the T16 decoders and tried to Verilate it standalone and it hit the memory limit.  I then started to reduce the number of case statements so see if an of the functions were causing the issue but it just seems to be the total number of these casez items that cause the issue.  Going from 4 to 8 shows a increase in processing time and memory usage and then past 8 things start to fall down.  I'm guessing expanding all these `?` is a state space explosion.  I have just written a Python script to expand all the casez items and I get a 49GB output file.  :-)  I guess I see the issue.

wsnyder, 2022-05-16:
Verilator shouldn't be expanding the ?'s beyond 2^16th in order to prevent the sort of memory problems you suspect.  Can you send the much smaller file as a standalone example - or much better keep digging in to debug it?

shareefj, 2022-05-17:
I can't currently send you the code as it's Arm copyright.  I'll see if I can get permission.

In the meantime, you were right that `-Ox` fixes the issue.  I've attached the log file that was generated when running with `--debug --debugi-V3Expand 9`.  The final two `.tree` files that it writes out are both 1.5GB if that means anything.  It dies at that point.

[verilator.log.gz](https://github.com/verilator/verilator/files/8711955/verilator.log.gz)

shareefj, 2022-07-22:
@wsnyder Just in case that last log file didn't help, here's a testcase that causes Verilator to chew all available memory.  Tested with v4.224

[testcase.zip](https://github.com/verilator/verilator/files/9168287/testcase.zip)

wsnyder, 2022-07-28:
The code you sent doesn't compile.

shareefj, 2022-07-28:
Sorry, trying again..
[testcase.zip](https://github.com/verilator/verilator/files/9214625/testcase.zip)

wsnyder, 2022-07-31:
The code is using wide arrays to "emulate" strings as it's pre-SV code, and they are in some large loops. This code is unrolled and becomes large, then V3Expand tries to clean things up and it needs A LOT of memory to do so.  There's no hang if you give it enough memory, I suspect your system is paging.

Use -fno-expand for the time being to get around this.

It's not clear what heuristics need tuning to make this better, probably we shouldn't unroll the loops, but this may cause performance problems or UNOPTFLAT warnings in existing designs..

shareefj, 2022-08-08:
OK, I tried that and it doesn't help much.  The memory usage is still bloating to 20GB and this model needs to be run on laptops with 8GB.  I've also just killed the build after around 5 mins or more where the usual build time for this model is seconds.

Would it be worthwhile converting the code to SV and using strings?

wsnyder, 2022-08-08:
>Would it be worthwhile converting the code to SV and using strings?

That would almost certainly solve the problem.

shareefj, 2022-08-08:
Doesn't seem to help much.  I'm still getting high memory usage (more than end users can tollerate) and I've given up waiting for the build to finish.  I'll park this for now and watch this issue for possible future updates.

wsnyder, 2026-11-13:
User is no longer on github so assuming not important, closing.

closed 2026-11-13 by wsnyder