verilator/verilator #3418: Publish versioned documentation

gezalore, 2022-05-15:
We agreed versioned documentation would be a good thing, so we don't necessarily have to carry forward information about removed features, etc. This issue is for discussion and to track progress.

gezalore, 2022-07-12:
I'm not sure what you just said fits with what I thought we are discussing here.

My impression was that versioned documentation means we would have a separate documentation build for each release, not just for branches, e.g.: we would have (v4.x, v4.y, v4.z, then v5), and possibly an alias for 'latest', and an extra for head of master. Once a release is made, the docs for that don't change (and ideally the docs are built by CI anyway), so I don't know what would need "backporting".

As of people not realising they need to read the docs for the version they are actually using, I think we can expect sufficient intelligence from the user to do this, both gcc and llvm/clang does it this way: https://gcc.gnu.org/onlinedocs/

wsnyder, 2022-11-19:
Might happen at some point, but for now just tracking the latest stable, and we comment on flags that change.

closed 2022-11-19 by wsnyder