verilator/verilator #3419: Encapsulated verilated models independent of each other and the process

gezalore, 2022-05-15:
For v5, I would like to move toward a usage model and interface for verilated model that allow embedding them in an application in an arbitrary way. What I mean here, is that I would like to be able to instantiate a verilated model, interact with it, possibly crash it via $stop, and then be able to do the same again without having to re-start the application process, and also to do the same in parallel on multiple models.

This is currently hindered by at least the following:

- Reliance on the global `Verilated` namespace, an generally on any other global functions and variables. I believe this can be replaced by `VerilatedContext` that is simulation instance specific.

- Use of static data in some of the runtime library. This is similar to the above, but relates to implementations using static data. I think we can just substitute this with thread_local storage which should be safe

- Calling abort(), or otherwise terminating the process when a simulation error occurs. I propose we throw exceptions instead that the client can handle without terminating the process.

I would appreciate any thoughts.

wsnyder, 2022-05-15:
VerilatedContext was intended to solve the base static problems you describe.  Those remaining statics (other than Verilated itself which isn't needed if using VerilatedContext) should be able to be stomped out.

Throwing exceptions always scares me, especially in multithreaded code, I can probably get over it.  An alternative is to allow the user to register a different abort() function via VerilatedContext.

Are you suggesting we should get rid of Verilated:: entirely and force use of VerilatedContext?  I'm not sure if we want to force all that old code to break.

gezalore, 2022-05-15:
> Throwing exceptions always scares me, especially in multithreaded code, I can probably get over it. An alternative is to allow the user to register a different abort() function via VerilatedContext.

In either case, the thread pool need to be shut down properly, so that is something we will need to address. BTW, yes, being able to register custom handlers for things like vl_fatal and friends by overriding VerilatedContext methods is also something I would favour in contrast to the current ifdefery.

> Are you suggesting we should get rid of Verilated:: entirely and force use of VerilatedContext? I'm not sure if we want to force all that old code to break.

I would prefer to get rid of it, and FWIW replacing the use of Verilated:: with context-> is not much harder than running a sed expression. Something we could do is get rid of it by default, but add it back conditionally under a `VL_I_WANT_NEW_FEATURES_BUT_IM_NOT_ALLOWED_TO_FIX_MY_CODE` define, if someone really really can't or don't want to fix their code. That's probably not too much burden on a major version upgrade?

wsnyder, 2022-05-15:
>BTW, yes, being able to register custom handlers

Filed #3423.

>Something we could do is get rid of it by default, but add it back conditionally 

This is what -DVL_NO_LEGACY does, though the inverted default as I prefer backwards compatibility "working".

gezalore, 2022-05-15:
> This is what -DVL_NO_LEGACY does, though the inverted default as I prefer backwards compatibility "working".

See, I never knew that existed, even though I have read the manual back and forth multiple times and dig around in the guts regularly. By having this default off, it is not really discoverable.

wsnyder, 2025-09-26:
We've made general improvements in independent models over the years, closing as this seems stale. If/when you are ready to make more improvements let's discuss the specific sub-improvement in a new pull or issue, thanks.

closed 2025-09-26 by wsnyder