verilator/verilator #3420: Clean up Ast representations of statements and expressions

gezalore, 2022-05-15:
I have found myself on occasion struggling with conservative optimizations due to not knowing what AstNodes I need to be aware of at what point (and I have a couple patches parked for this reason). One thing that I bump into particularly often is the representations of procedural statements, and expressions. Most procedural statements derive from `AstNodeStmt`, and most expressions derive from `AstNodeMath`, but this is not universally true. `AstCCall` and related are being the prime examples, but probably there are other.

I would like to get to a point where all procedural statements (i.e.: those things that represent steps in the sequential execution of some program, and are executed for their _effect_) derive from `AstNodeStmt`, and all expressions (i.e.: those thing that represent a value [or LValue]) to derive from `AstNodeMath`. This would make it pretty clear what needs to be handled by various transformations.

This would also allow us to tighten some of the types in AstNode interfaces as suggested in #3264.

wsnyder, 2022-05-15:
This has also bothered me for a while, but I hadn't yet come up with a great solution.

Replicating AstCCall and all other node types that need to be either statements or expressions might end up error prone in a different direction.

Is AstVar a statement?  It can be in a statement position in some cases.  Perhaps make a list of all those that are "incorrect"?

A somewhat related problem is the mess to track where to insert more statements, e.g. for While loops it is complicated, see the code in e.g. V3LinkInc/V3Task and elsewhere.

One idea I've considered (but done nothing on) is to have a AstStatements node type, and any statement MUST be underneath it.  Somewhat like AstBegin but required to be present in every appropriate place. This then makes it easy to know where to insert, and likewise easy to know what is a statement (by following backp()).

gezalore, 2022-05-15:
>Replicating AstCCall and all other node types that need to be either statements or expressions might end up error prone in a different direction.
>
>Is AstVar a statement? It can be in a statement position in some cases. Perhaps make a list of all those that are "incorrect"?

These are actually simple: expressions in statement position would be AstStmtMath, which hold the AstNodeMath as child (emitting it just emitts the expression and appends a semicolon). Similarly we could have AstStmtDecl, an other similar 'splice' nodes. This is how I have done it in [Alogic](https://github.com/alogic/alogic/blob/6f3a0b958f4b7d18d1d40ddafb773e1983bf1c43/src/main/scala/com/argondesign/alogic/ast/Trees.scala#L286) and worked very well there.

wsnyder, 2022-11-19:
Finished with the commits above, I believe.

gezalore, 2022-11-20:
I have a few pending patches left for this

closed 2023-01-21 by wsnyder