verilator/verilator #3421: Fix develop-v5 extended tests DEPRECATED failure

wsnyder, 2022-05-15:
@gezalore 

Running the extended tests on develop-v5 branch, and updating that branch's verilator submodule to develop-v5 HEAD fails.

```
%Warning-DEPRECATED: /svaha/wsnyder/SandBox/homecvs/v4/verilator_ext_tests/t/../submodules/Cores-SweRV-EH2/design/lib/beh_lib.$
 1223 |    logic  en_ff /*verilator clock_enable*/ ;
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

I think we had agreed that this would be ignored and not deprecated for backward compatibility. Note the code in question it outside our control, and still needs to work with V4, so is a good example of why ignoring this is a arguably better experience.

When fixed, please consider pushing the submodule update.

gezalore, 2022-05-15:
> Running the extended tests on develop-v5 branch, and updating that branch's verilator submodule to develop-v5 HEAD fails.

Sorry, I don't understand what 'updating that branch's verilator submodule to develop-v5 HEAD' means here. It all passed on my checkout.

gezalore, 2022-05-15:
I wonder if it would be helpful to add a new class of message (call it 'note') that is non-fatal by default and issue it on these no-effect things as it will make people wonder. We could then
- Use DEPRECATED (warning) that will result in breakage in the future and will need to be fixed (do we have much of these actually, other than removal of obscure features? Like at some point this was used for changing command line/config file syntax, but you seem to be against any sort of breakage these days)
- Add a NOEFFECT (note) that is accepted but ignored, and will not fail verilation even with -Wfatal. This means old code works but there is a little encouragement to move on with the world for projects that can afford to.

wsnyder, 2022-05-15:
Probably easiest is to make a pull request against https://github.com/verilator/verilator_ext_tests/tree/develop-v5 to change the submodule for verilator to develop-v5 head.

Or, to run manually, roughly:

```
git clone git@github.com:verilator/verilator_ext_tests.git
cd verilator_ext_tests/
git submodule update --recursive --init
git checkout develop-v5
git submodule update --remote --merge
( cd submodules/verilator && autoconf && ./configure && make)
export VERILATOR_ROOT=`pwd`/submodules/verilator
make test
```

If that gets to pass, you can then "git push" it.

wsnyder, 2022-05-15:
I view DEPRECATED as "what you are doing works now but in the future will break".  An example is the --cdc feature.  Having a /*verilator clock_enable*/ is not in that category, as in the future ignoring it will cause no harm, that is not break the user's expectations.

gezalore, 2022-05-15:
> I view DEPRECATED as "what you are doing works now but in the future will break". An example is the --cdc feature. Having a /_verilator clock_enable_/ is not in that category, as in the future ignoring it will cause no harm, that is not break the user's expectations.

On this we are in agreement. I'm just confused that you are happy to deprecate some things in a non-backward compatible way, while not others :) (example was the -rule option in config files)

gezalore, 2022-05-15:
> Probably easiest is to make a pull request against https://github.com/verilator/verilator_ext_tests/tree/develop-v5 to change the submodule for verilator to develop-v5 head.
>
> Or, to run manually, [...]

So I think I done this, except I pointed VERILATOR_ROOT at my development checkout, instead of into the submodule

gezalore, 2022-05-16:
Ah, apparently I added a -Wno-DEPRECATED to the EH2 test in my local checkout :sweat_smile:

gezalore, 2022-05-16:
@wsnyder if you set something to status:assigned, it would be helpful to also add the person as the Assignees for easier sorting

commit 0e62cd11 (2022-05-16): Don't issue DEPRECATED for now no-op clock_enable attribute

Fixes #3421
closed 2022-05-16 by gezalore