verilator/verilator #3422: Check develop-v5 code coverage

wsnyder, 2022-05-15:
Please look through this report and see if any new code that is uncovered (red/yellow on left) that you feel needs additional testing or otherwise to get covered.  

    https://codecov.io/github/verilator/verilator/commit/599d23697dabe2da7dcd6e2cd863c1f9adec8e41

Note you can run this manually locally with

nodist/code_coverage

then look at the HTML files generated.  It doesn't post to codecov automatically but rather prints the command to do it.

It in general the coverage seems extremely clean (much better than I usually get first run) so perhaps you already did this.  No need to get to 100%, the src is at about 88% FWIW.

wsnyder, 2022-05-15:
@gezalore

gezalore, 2022-05-16:
Mostly dead code that is removed. The branch coverage is very noisy :/

commit 282887d9 (2022-05-16): Fix code coverage holes

Fixes #3422
closed 2022-05-16 by gezalore