verilator/verilator #3424: Remove VL_THREAD_LOCAL

gezalore, 2022-05-15:
`thread_local` is a C++11 standard keyword, do we till need `VL_THREAD_LOCAL` now that we require at least C++11?

wsnyder, 2022-05-15:
VL_THREAD_LOCAL isn't thread_local when not using threads, which makes it smaller and faster.

I just checked on a recent GCC and this is still true.

```
extern "C" int calct();
int calct() {
    static thread_local int t_v = 12;
    return t_v++;
}
extern "C" int calcs();
int calcs() {
    static int t_v = 12;
    return t_v++;
}

0000000000000000 <calct>:
   0:   f3 0f 1e fa             endbr64
   4:   64 8b 04 25 00 00 00    mov    %fs:0x0,%eax
   b:   00
   c:   8d 50 01                lea    0x1(%rax),%edx
   f:   64 89 14 25 00 00 00    mov    %edx,%fs:0x0
  16:   00
  17:   c3                      retq

0000000000000000 <calcs>:
   0:   f3 0f 1e fa             endbr64
   4:   8b 05 00 00 00 00       mov    0x0(%rip),%eax        # a <calcs+0xa>
   a:   8d 50 01                lea    0x1(%rax),%edx
   d:   89 15 00 00 00 00       mov    %edx,0x0(%rip)        # 13 <calcs+0x13>

```

gezalore, 2022-05-15:
Hmm, I missed that, but that would still violate #3419 (running two single threaded models at the same time on different host threads in the same app) ?

wsnyder, 2022-05-15:
VL_THREADED must be set in that case, as thread safety is required, --threads 1 must be used then.

closed 2022-05-16 by gezalore