verilator/verilator #3433: Localparam integer array is not constant when initialized with slice of larger localparam integer array

michael-platzer, 2022-05-17:
Hi,

when initializing a localparam integer array with a slice of a larger localparam integer array, then Verilator complains that the resulting localparam is not constant. These arrays are initialized as follows:
```
localparam int unsigned LARGE_ARRAY[5] = '{1, 2, 3, 4, 5};
localparam int unsigned SMALL_ARRAY[2] = LARGE_ARRAY[1 +: 2];
```

As far as I understand the standard, slices of unpacked arrays are permissible and by extension slices of constant integer arrays should equally be legal in SystemVerilog (and be constant as long as the source array and slice indices are all constant).

Code example:
```
module mymod
    #(
        parameter int unsigned VAL[2] = '{1, 2}
    )
    ();
endmodule

module top
    ();
    localparam int unsigned LARGE_ARRAY[5] = '{1, 2, 3, 4, 5};
    localparam int unsigned SMALL_ARRAY[2] = LARGE_ARRAY[1 +: 2];
    mymod #(
        .VAL ( SMALL_ARRAY )
    ) myinst ();
endmodule
```

Verilator version and error message:
```
$ verilator --version
Verilator 4.222 2022-05-02 rev UNKNOWN.REV
$ verilator -Wall --cc array_params.sv --top-module top --lint-only
%Error: array_params.sv:13:16: Expecting expression to be constant, but variable isn't const: 'SMALL_ARRAY'
                                       : ... In instance top
   13 |         .VAL ( SMALL_ARRAY )
      |                ^~~~~~~~~~~
%Error: array_params.sv:13:10: Can't convert defparam value to constant: Param 'VAL' of 'myinst'
                                       : ... In instance top
   13 |         .VAL ( SMALL_ARRAY )
      |          ^~~
```

Note that directly assigning the desired values to the second array as shown below works fine.
```
localparam int unsigned LARGE_ARRAY[5] = '{1, 2, 3, 4, 5};
localparam int unsigned SMALL_ARRAY[2] = '{2, 3};
```

michael-platzer, 2022-05-19:
Btw, this issue also affects the current v5 beta:

```
$ ./bin/verilator --version
Verilator 5.001 devel rev v4.222-28-g282887d9
$ ./bin/verilator -Wall --cc ../array_params.sv --top-module top --lint-only
%Error: ../array_params.sv:13:16: Expecting expression to be constant, but variable isn't const: 'SMALL_ARRAY'
                                : ... In instance top
   13 |         .VAL ( SMALL_ARRAY )
      |                ^~~~~~~~~~~
%Error: ../array_params.sv:13:10: Can't convert defparam value to constant: Param 'VAL' of 'myinst'
                                : ... In instance top
   13 |         .VAL ( SMALL_ARRAY )
      |          ^~~
```

wsnyder, 2022-05-20:
Slice handling in constant code is currently not supported.

If you'd like to work on this it would be appreciated, this is otherwise likely to be a while before fixed.  Basically the slice handling code in V3Width currently converts directly to runtime-only function calls, rather than AstNode types that V3Const/V3Simulate could handle.