verilator/verilator #3436: Move to GCC style -f options for optimizations

gezalore, 2022-05-17:
The current `-O{a,b,c,d,e,g,i,k,l,m,o,p,r,s,t,u,v,w,x,y,z}` (yes, those are all the letters) is pretty untractable (and I have a few optimizations pending that will make us run out of letters eventually), I suggest deprecating these and using `-f<longname>` instead. We do have a bare `-f` options but that is hopefully not ambiguous.

wsnyder, 2022-05-17:
We could use a letter to escape the next letter, but I think using -f is a fine plan.

gezalore, 2022-05-17:
`-OoOoo`

wsnyder, 2022-06-03:
@gezalore let me know if you start on this.

gezalore, 2022-06-03:
I have not started and might not for a little longer, in case you want to tackle this.

One nit about this is that with our current option scheme it would be `-ffoo` and `-no-ffoo`, which is unfortunate and we should probably stick to the C compilers style `-fno-foo`.

Note to implementer: The `--merge-const-pool` option should probably also be converted.

gezalore, 2022-06-04:
Thank you for handling this. I'm reopening as reading the patch the documentation seems to mention `--O*` and `--f*` options. Think the intention, and implementation is that these are actually single `-` options? The `-O*` definitely are listed with a single `-` in the `bin/verilator` script. Depending on what the intention was I think these should be consistent.

wsnyder, 2022-06-04:
The parser doesn't care about -- or - on any option. Generally GCC/standard verilog usually have single -, others double.  But agree docs should be consistent, fixed.

closed 2022-06-04 by wsnyder