verilator/verilator #3437: Remove V3Expand, use VlWide methods instead

gezalore, 2022-05-17:
I think a bulk of the functionality of `V3Expand` could be implemented by appropriate operators and methods over `VlWide` (similar to some of the macros), which could then remain as loop based library routines (and include intrinsics if needed). This would likely reduce code size and Improve performance.

wsnyder, 2022-05-17:
V3Expand didn't use to exist and the compiler did what you suggest which is why all the _W functions are there but sometimes not used. (Of course using VlWide makes the code more readable, but results in similar final assembly to the _W's). V3Expand was specifically added because it reduced compiler time by 30% or more and runtime by about 10%.  Compilers have gotten better but I think it's unlikely they'll do as well. Of course I welcome data proving otherwise - it's easy to disable to try.

gezalore, 2022-05-20:
Indeed disabling V3Expand as is hurts performance, but I think there is more to the story so will investigate more.

wsnyder, 2022-08-30:
I'm pretty confident V3Expand is likely to remain valuable, if you provide evidence otherwise feel free to reopen.

closed 2022-08-30 by wsnyder