verilator/verilator #3438: How to address - "Unsupported: Timing control statement in this location"

phelter, 2022-05-18:
Have the following code:
```verilog

interface x_bus #(ADDR_WIDTH=32) (input wire clk);
    `LOCAL_KEEP logic [ADDR_WIDTH-1:0] address;
    `LOCAL_KEEP logic [31:0]           write_data;
    `LOCAL_KEEP logic [31:0]           read_data;
    `LOCAL_KEEP logic                     write;
    `LOCAL_KEEP logic                     read;
    `LOCAL_KEEP logic                     write_done;
    `LOCAL_KEEP logic                     read_done;

    modport slave (input address, write_data, write, read, output read_data, write_done, read_done);
    modport master  (input read_data, write_done, read_done, output address, write_data, write, read );

    task write_reg(input string name, input integer addr, input integer value);
        integer timeout = 256;
        begin
            @(posedge clk);
            timeout    = 256;
            address    = ADDR_WIDTH'(addr);
            write_data = value;
            write      = 1;
            read       = 0;
            while( !write_done) begin
                @(posedge clk);
                write = 0;
                if (timeout > 0) timeout = timeout-1;
                else begin
                    $error("%s: timeout in OB write", name);
                    break;
                end
            end
            address    = 0;
            write_data = 0;
        end
    endtask
endinterface
```

The `@(posedge clk)` seem to be causing these errors in verilator.

The error message:
```text
%Error-UNSUPPORTED: <filename>:60:13: Unsupported: timing control statement in this location
                                        : ... In instance X
                                        : ... Suggest have one timing control statement per procedure, at the top of the procedure
```

How would one go about having `one timing control statement` per procedure at the top of the procedure/task? in this case?

wsnyder, 2022-05-20:
Verilator v4 doesn't have dynamic timing controls, you'd need to write this as a FSM with some state variable that is checked/updated in an always@posedge block rather than a function.

There's some experimental work towards v5 but that code isn't yet committed see #3363 if you want to help out or try it.

wsnyder, 2022-05-20:
Closing as answered, we know this isn't supported so not much reason to keep it open.  If you have additional questions feel free to follow up here.

closed 2022-05-20 by wsnyder