verilator/verilator #3440: struct parametrization via virtual class parameter

solomatnikov, 2022-05-21:
Thanks for taking the time to report this.

What would you like added/supported?

**struct parametrization via virtual class parameter**

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

Simplifies example from http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf

```
virtual class C#(parameter type T = logic, parameter SIZE = 1);
   typedef logic [SIZE-1:0] t_vector;
   typedef T t_array [SIZE-1:0];
   typedef struct {
      t_vector m0;
      t_array m1;
   } t_struct;
endclass

module test_top ();
   typedef logic [7:0] t_t0;
   C#(t_t0,3)::t_vector v0; C#(t_t0,3)::t_array a0;
   C#(bit,4)::t_struct s0;
endmodule
```

```
%Error-UNSUPPORTED: /home/sols/work/wss/rtl/tl_pkg.sv:108:54: Unsupported: class parameters
                                                            : ... In instance top
  108 | virtual class C#(parameter type T = logic, parameter SIZE = 1);
      |                                                      ^~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=4.203
```

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2022-05-21:
This is currently unsupported; duplicate of #2231.  If you want to help on that one let us know there please.

closed 2022-05-21 by wsnyder

solomatnikov, 2022-05-22:
This is much more limited in scope than #2231, i.e. not general class parameterization, just struct parameterization inside the class because direct struct parameterization is not possible in SystemVerilog. 

Simple parameters, without type parameters, would be sufficient.

wsnyder, 2022-05-22:
I think the general fix comes along with this for nearly free.

solomatnikov, 2022-05-22:
> I think the general fix comes along with this for nearly free.

Ok, what needs to be done for this?