verilator/verilator #3441: typedef derived from type defined inside interface

solomatnikov, 2022-05-22:
Thanks for taking the time to report this.

What would you like added/supported?

**typedef derived from type defined inside interface**

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

https://verificationacademy.com/forums/systemverilog/parameterized-struct-systemverilog-design#answer-61637

```
interface common_intf #(int  ADDR_W, DATA_W);
typedef struct packed {

logic [ADDR_W-1:0] cntr_idx;
logic [DATA_W-1:0] cntr_val;

} stage_t;

  stage_t bus;

endinterface

module mod1(common_intf p1);
  initial $display("%m idx bits %0d, val bits %0d", $bits(p1.bus.cntr_idx), $bits(p1.bus.cntr_val));
  typedef p1.stage_t stage_t; // "imports" type into module
  stage_t local_bus;
endmodule

module mod2(common_intf p1, p2);
  initial $display("%m-1 idx bits %0d, val bits %0d",
                   $bits(p1.bus.cntr_idx),
                   $bits(p1.bus.cntr_val));
  typedef p1.stage_t stage1_t; // "imports" type into module
  stage1_t local_bus1;
  initial $display("%m-2 idx bits %0d, val bits %0d",
                   $bits(p2.bus.cntr_idx),
                   $bits(p2.bus.cntr_val));
  typedef p2.stage_t stage2_t; // "imports" type into module
  stage2_t local_bus2;
  initial $display("%m params p1 ADDR_W %0d DATA_W %0d", p1.ADDR_W, p1.DATA_W);
  initial $display("%m params p2 ADDR_W %0d DATA_W %0d", p2.ADDR_W, p2.DATA_W);

endmodule

module top;
  common_intf #(8,16) i1_2(); // connects m1 to m2
  common_intf #(4,32) i2_3(); // connects m2 to m3
  mod1 m1(i1_2);
  mod2 m2(i1_2,i2_3);
  mod1 m3(i2_3);
endmodule
```

```
%Error-UNSUPPORTED: ....sv:123:3: Unsupported: SystemVerilog 2005 typedef in this context
  123 |   typedef p1.stage_t stage_t;  
      |   ^~~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=4.221
%Error-UNSUPPORTED: ....sv:131:3: Unsupported: SystemVerilog 2005 typedef in this context
  131 |   typedef p1.stage_t stage1_t;  
      |   ^~~~~~~
%Error-UNSUPPORTED: ....sv:136:3: Unsupported: SystemVerilog 2005 typedef in this context
  136 |   typedef p2.stage_t stage2_t;  
      |   ^~~~~~~
```

May we assist you in trying to fix this in Verilator yourself?

goekce, 2022-12-29:
Related: #2783

solomatnikov, 2023-06-02:
It turns out something similar can be done with classes, .e.g.:

```
virtual class C#(parameter type T = logic, parameter SIZE = 1);
    typedef logic [SIZE-1:0] t_vector;
    typedef T t_array [SIZE-1:0];
    typedef struct {
        t_vector m0 [2*SIZE-1:0];
        t_array m1;
    } t_struct;
endclass

module t_io_vclass_typedef #(parameter type T = logic, parameter SIZE = 1) (
   // Inputs
   input  C#(T,SIZE)::t_struct in_struct,
   // Outputs
   output C#(T,SIZE)::t_vector out_vector,
   output C#(T,SIZE)::t_array  out_array
   );

   assign out_vector = in_struct.m0[0] + in_struct.m0[1];
//   assign out_array  = in_struct.m1;

endmodule

module t (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   C#()::t_struct c_struct1;
   C#()::t_vector c_vector1;
   C#()::t_array  c_array1;

   typedef logic [3:0] c_mod2_t;
   C#(c_mod2_t,2)::t_struct c_struct2;
//   C#(c_mod2_t,2)::t_vector c_vector2;
   C#(c_mod2_t,2)::t_array  c_array2;

   typedef C#(c_mod2_t,2)::t_vector c_vector2_t;
   c_vector2_t c_vector2;

   assign c_struct1.m0[0] = 1'd0;
   assign c_struct1.m0[1] = 1'd1;
   assign c_struct1.m1[0] = 1'd1;

   assign c_struct2.m0[0] = 2'd1;
   assign c_struct2.m0[1] = 2'd2;
   assign c_struct2.m1[0] = 4'd3;

   t_io_vclass_typedef i_t_io_vclass_typedef1(.in_struct (c_struct1),
				              .out_vector(c_vector1),
					      .out_array (c_array1));

   t_io_vclass_typedef #(c_mod2_t,2) i_t_io_vclass_typedef2(.in_struct (c_struct2),
							    .out_vector(c_vector2),
							    .out_array (c_array2));
```

But it still doesn't work because of #4256

jeras, 2023-08-19:
I often use Vivado for reference, since it is free to use and has a rather good SystemVerilog support.
I am mentioning this here, since it is an indicator of how relevant a feature is.

So Vivado 2023.1 synthesis supports access to types inside interfaces, but the simulator doesn't:
```
ERROR: [VRFC 10-1753] hierarchical name in type reference is not allowed [/home/ijeras/VLSI/rp32/submodules/tcb/hdl/rtl/lib/tcb_lib_demultiplexer.sv:57]
```

jeras, 2023-08-21:
And I also tested with the latest Intel Questa FPGA Starter Edition and I get:
```
Error: .../rp32/submodules/tcb/hdl/rtl/lib/tcb_lib_demultiplexer.sv(57): 'sub' is an unknown type.
```
And since I exhausted my free simulator options, I will revert the RTL to a version where I redefine the type locally.

Someday I will have to run a netlist simulation on the Vivado synthesis output to see if it actually interpreted the code correctly.

em2machine, 2025-11-24:
@wsnyder  I don't know if its interesting / useful, but I've been working on a PR for #6637 (PR 6732) that seems in similar vein.  (localparam X_t = if0.X_t; versus the typdef form)....

jeras, 2025-12-06:
I will test my code when this feature is implemented.

There is another note I would like to add. If the type is used directly without an intermediate `typedef`, Verilator crashes, Questa would return a warning (something like "not a type").

Correct code:
```
typedef sysbus.request_t request_t;
request_t req;
```
Code causing crash (should be error):
```
sysbus.request_t req;
```