verilator/verilator #3442: --debug segfault on develop-v5 on BlackParrot

dpetrisko, 2022-05-22:
Hi, I am seeing a regression on develop-v5. Curiously, the actual build and execution of a simulation works fine. However, using --debug to try to debug an unrelated issue causes a segfault during verilation. Unfortunately, I don't have time currently to shrink the testcase

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

To reproduce:

```
git clone -b dev git@github.com:black-parrot/black-parrot
cd black-parrot
git submodule update --init external/
# Apply this patch

diff --git a/bp_common/syn/Makefile.verilator b/bp_common/syn/Makefile.verilator
index 50154a31..6cba9d59 100644
--- a/bp_common/syn/Makefile.verilator
+++ b/bp_common/syn/Makefile.verilator
@@ -14,7 +14,7 @@ VV ?= $(VERILATOR)
 VERILATOR_THREADS ?= 1
 ## Tool options
 VV_OPTS  = --cc                 # Output in C++
-#VV_OPTS += --debug --gdbbt     # Debugs and produces stack trace
+VV_OPTS += --debug --gdbbt     # Debugs and produces stack trace
 VV_OPTS += -O1
 VV_OPTS += --x-assign fast --x-initial fast
 VV_OPTS += --threads $(VERILATOR_THREADS)

make -C bp_top/syn clean build.sc

Result:
%Error: Internal Error: ../V3OrderGraph.h:84: Must not be null
-vertex: 0x5555cf910630
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1108:     Dumping obj_dir/Vtestbench_990_final.tree
- V3StatsReport.cpp:220:statsReport: 
%Error: Internal Error: Aborting since under --debug

```

What 'verilator --version' are you using?  Did you try it with the git master version?

develop-v5

What OS and distribution are you using?

Ubuntu 20.04

May we assist you in trying to fix this in Verilator yourself?

I'm happy to try out solutions, but have little time to debug myself for the next month or so

gezalore, 2022-05-22:
Thanks for the report. Glad to hear the non-debug build is actually fine. I will fix in the near future but I am assuming it's not otherwise urgent, let me know otherwise.

Note to self: head commit at reporting time: 6c8e2f871223e5aa16a825e94a1c168835eddb88

kbieganski, 2022-05-25:
I believe this is caused by a bug already present in `master`. Specifically, this line:
https://github.com/verilator/verilator/blob/a372e010bda869f0768c2de86729ea2103001a87/src/V3Order.cpp#L378

`getVarVertex` constructs an `OrderVarVertex` using `m_scopep`, which is uninitialized at this point (class members in C++ are initialized in the order they appear in the class, and `m_scopep` is further down). Previously it didn't cause issues, so I guess the scope isn't used for this vertex. However, in `develop-v5` there's a new assert:
https://github.com/verilator/verilator/blob/f0a2c983767835ce74e042f08a60c7982951bcd1/src/V3OrderGraph.h#L84
Our scope pointer can contain anything at this time, but sometimes it happens to be a `nullptr`, triggering this assert.

Whether it manifests or not is quite random and depends on the compiler, whether it's the debug build, and probably other things. I encountered this after merging `master` into `develop-v5`: https://github.com/antmicro/verilator-1/actions/runs/2370749796

Not sure how to fix it without removing the assert, so I'll leave it up to you @gezalore.

gezalore, 2022-05-25:
Thanks for tracking this down, I think you are right. I was planning on removing the DPI vertex and handling it via the 'extraDomains' closure passed to order. In the meantime you can disable the assert if it causes problems, everything else should have a valid scope

gezalore, 2022-05-25:
Now fixed on develop-v5 by moving the scope pointer and assertion to `OrderLogicVertex`, which is the only place it should matter.

closed 2022-05-25 by gezalore

dpetrisko, 2022-05-25:
Thanks for the quick fix!

commit 3af5e7e8 (2022-05-25): Remove scope pointer from OrderEitherVertex.

For ordering, only the scope of logic vertices should be relevant, so
remove the scope pointer from OrderEitherVertex and move it into
OrderLogicVertex. This does not change single-threaded scheduling at
all. Theoretically, multi-threaded scheduling should not be affected
either though due to some implementation quirk depending on vertex order
in a graph the MT schedule is perturbed by this change, but the
performance effect of this is negligible on all benchmarks I have access
to.

No functional change intended.

Fixes #3442