verilator/verilator #3447: What can I do if I want to add a static lib into verilator

specialpointcentral, 2022-05-27:
Recently, I want to add my static lib into verilator. So I use `<file.a>` to add into verilator args. It works.
But what confused me is verilator self-generate makefile did not give this lib's dependency rules.
So if I modify my lib files and compile them to `file.a`, verilator thinks this lib is ok, and does not recompile the `VTop`.
What can I do to deal this problem.

Thanks.

wsnyder, 2022-05-27:
You'll need to make your own makefile that has a dependency on the final executable, then includes the generated makefile, e.g.

```
include Vtop.mk
mylibrary.a: {.... whatever rules are needed}
Vtop: mylibrary.a
```

closed 2022-05-27 by wsnyder

specialpointcentral, 2022-05-29:
That works, Thank a lot.

> You'll need to make your own makefile that has a dependency on the final executable, then includes the generated makefile, e.g.
> 
> ```
> include Vtop.mk
> mylibrary.a: {.... whatever rules are needed}
> Vtop: mylibrary.a
> ```