verilator/verilator #3448: Unexpected assertion behaviour

tbialas-riverlane, 2022-05-27:
Dear Verilator team,

I have run into an issue that I am unsure how to resolve.
I have a design containing signal stability assertions, of the form:

`assert property (@(posedge clk) disable iff (~reset_n) ~rval | (raddr0_r==addr0_i));`

Sometimes, with minor changes to the design not affecting the signals in the property, the assertions will start to fail exactly at time 10000 when Verilator is launched with `--assert` enabled on compilation and runtime.
Additionally, when compiled with additional trace options `--trace-fst --trace-structs` , the assertions do not fail. I am able to make them fail again by recompiling without the trace options.

Any guidance would be appreciated.
Thank you,
Best regards,
Tomasz

wsnyder, 2022-05-27:
Perhaps raddr_r or related signal come from async logic rather than a clock and so are nondetermistic in when their processes may be evaluated?  Add a print statement to see when they change.  Otherwise we'll need a self-contained minimal example to determine what's happening.

tbialas-riverlane, 2022-05-31:
That is correct, the signal comes from async logic; it seems order is nondeterministic in evaluation out of reset. I tested with QuestaSim and the assertions do not fail.
I will try to provide a self-contained minimal example. Thank you!

wsnyder, 2022-08-30:
Feel free to reopen if provide an example.

closed 2022-08-30 by wsnyder