verilator/verilator #3454: Move the thread pool into the context

gezalore, 2022-05-29:
Currently the thread pool is part of the model. This is unfortunate, as in multiple model simulations, the workers of one model will spin-wait while the other model executes, creating unnecessary contention.

I think the intention was to move the thread pool into the context, so I suggest we do so, and resize the thread-pool to accomodate the widest model, as models are added to the context (it is not possible to run a wider model on a narrower thread-pool as the worker indices are baked in).

gezalore, 2022-05-29:
@wsnyder would you like this to be v5 only or master too?

wsnyder, 2022-05-29:
Hmm, I think the current model is broken so would suggest lean to v4 unless it gets more intrusive than I suspect.  I'm open to alternative argument.

gezalore, 2022-06-13:
Ok, I'm running into another one of the horrors of having the run-time library built together with each model. (This time the execution profiler being present or not depending on defines.) I am marking this blocked as keeping all of: API backwards compatibility, functionality backwards compatibility, and making this change is pretty much intractable.

gezalore, 2022-06-21:
Note: In all the following "profiler" means the "execution profiler", it's another question what makes sense for the PGO profiler.

> For VL_PROFILER specifically I think this can easily be resolved by having both a startWorkerProfiler and startWorker, and having Verilator select which at verilation time. Then selectively compile in verilated_profiler.

I don't quite understand how that works.The problem is, when creating the thread (or at least before running any model code on it), we need to initialise the per thread profiler data structures. If we want the thread pool to be in the context, then the profiler should be in the context as well, otherwise we could have multiple profilers associated with each thread (As a side note: I think we should have the execution profiler - or at least it's current incarnation - in the context anyway, as profiling the whole simulation - i.e.: multiple models together - makes sense and works naturally if the thread pool is shared). If the thread pool is shared, you need to be able to up-size it to the widest model when that model is added to the context. Now if you have a model compiled without `--prof-exec` and another one with `--prof-exec`, and then you add the non-profiled model to the context first, then the profiled one, you need to ensure the per thread profiler data is set up on the threads created by the non-profiled model. We need to either initialize the profiler data after thread creation (which is doable, and we need to do this anyway I think), or defer thread creation until after all models have been added to the context and we know whether we use the profiler. I don't see how the suggestion solves this.

> I think this can easily be resolved by having both a startWorkerProfiler and startWorker, and having Verilator select which at verilation time. Then selectively compile in verilated_profiler.

I think you will find that startWorkerProfiler (which lives in verilated_thread) will not link without verilated_profiler, and hence would need to live under an ifdef VL_PROFILER (see the comment in the ifdef in the current `startWorker`), or be moved to verilated_profiler, neither of which are very clean. 

In general we always need some state in the thread pool too keep track of profiler initialization, this could be as simple as a `bool` by moving all initialisation related code into verilated_profielr, but it just makes the run-time library more opaque, which I think is a maintainability problem.

The only way of solving this that I currently see, that is both sound and admits a straight forward implementation (that is: without even more ifdefs or dances around the boundaries of the runtime due to parts of the runtime only being available in certain conditions) is to make the profiler always available, and decide at run-time whether we want to use it (i.e.: remove the VL_PROFILER defines). This would require always including verilated_profiler in the link (in case it is used by any of the models in the link, but more importantly, to avoid ifdefing in other parts of the runtime, notably in verilated_threads). There is no run-time performance impact to doing this as it's all run-once initialisation code, and there is no significant build time impact either on single models (i.e.: user code), as we just include verilated_profiler all the time. I know this slows down testing as now very test needs to compile verilated_profiler, even though it's picked up from the ccache almost all the time.

> This is also a small step closer to a unified library without defines.

If it was me, I would just do this as I am quite reluctant to add more ifdefs to the runtime. Once we have this, we could also just build the run-time ahead of time once into a `libverilated.a` which we have discussed doing some time ago, which would eliminate the excess build time of the above. We could have a few flavours like `libverilated-st.a` (where thread local is no thread local) vs libverilated-mt.a (where thread local is thread local) builds if needs be, but with a link-time assert that we are using the one expected by the model.

But more generally, all this brings us back to #2359 and the compatibility problem. Namely that some people expect models built with different versions of Verilator to run with the runtime of another version of Verilator, which I think is pretty unreasonable (and I don't think it has ever worked reliably), hence my claim of having to give up at least one of:
- Soundness: The current model is somewhat broken, as we have established
- Version (internal API) compatibility: Would have to rework a lot of stuff to make it sound
- Maintainability: Keeping internal API backward compatibility would be pretty expensive in developer effort and promising forward compatibility would likely hinder future work

I know which way I would go, but I think our visions for the tool are somewhat different, maybe we should start by clarifying that :)

wsnyder, 2022-06-22:
I agree we should be aiming for the execution profiler being part of the context.

The changes I were suggesting were only an intrum one to avoid the ifdef, not the larger thread pool refactoring.

### First step

As an intrum step I still agree we want to remove the VL_PROFILER define.  Until we have a library I don't want to always compile
verilated_profiler, so I'll make an opaque void callback we pass into startWorker.  For non-profiled runs it'll be nullptr, which
is fine as that routine isn't fast path anyways.  As it's an opaque function pointer this won't require any verilated_profiler to be built in when unused.

### Longer term

For ThreadPool the profiler is the smaller part of the problem. Currently the thread pool is created at the same time as the model is constructed. Obviously that won't work with multiple models of different widths, etc.

So a strategy might be:

1. Change model constructor to not make a ThreadPool any longer.  Instead they indicate what they will need in terms of thread counts, profiling setup callbacks, etc, by storing this information into some new within-context data structure.

2. The first call to eval() (perhaps other places??) then asks the context to make the thread pool.  The first model that calls this
will make the thread pool using the information across all models in step 1.  This code would also do the profiling setup. Subsequent model's first eval()s would skip all that (perhaps besides some obvious assertion checks).

### Other notes

> But more generally, all this brings us back to #2359 and the compatibility problem. Namely that some people expect models built with different versions of Verilator to run with the runtime of [...]

In #2359 I was agreeing with your points  ;) I don't think anyone is currently operating under that assumption as it just wouldn't work.  I see this as unreasonable at least at present and should not be a requirement on us.

>We could have a few flavours like libverilated-st.a (where thread local is no thread local) vs libverilated-mt.a (where thread local is thread local), but with a link-time assert that we are using the one expected by the model.

Right.

I do still see it as necessary to avoid breaking people's old build environments, at least to the extent reasonable for those whom
were doing things rather ordinarily.  (On that point we might differ in opinion.) It should be possible for us to get to new make rules that build an e.g. _mt and _st library, and still retains the older rules for separate object compilation for those with older makefiles (which wouldn't be using the library).

Note we also have ifdefs in the tracing implementation that need cleaning up before we can have a common library.  Or maybe those translate to the _mt vs _st flavors.

gezalore, 2022-06-22:
Strategy wise: I would just create the [additnional] threads needed as models are added (so we don't need to worry about the 'parhaps other places??'), and when a profiled model is added, execute the profiler callback on all the pre-existing threads. This is all doable if there is only one profiler (which should then live in the context), so I think we should just move both.

> I do still see it as necessary to avoid breaking people's old build environments, at least to the extent reasonable for those whom
were doing things rather ordinarily. 

How do you define ordinary? Ordinary to me is invoking the generated makefile. which in the library world could just add -lverilated to the link line, rather than building the verilated*.cpp, which work stransparenty. Anything else I would consider extraordinary and should be OK to break. It should also not be very hard to fix ones own build: 1. Remove verilated_*.o object, 2. add -lverilated.

wsnyder, 2022-06-24:
For the unified thread pool how does the user control the separate verilated model's threads to the master pool?  You seem to suggest there's a mapping like this:

    modelA thread0 -> pool thread 0
    modelB thread0 -> pool thread 0
    modelC thread1 -> pool thread 1

Which I think is making an assumption that might not be best performing especially if the total machine CPU count is enough to support all the model's threads separately (e.g. 3 above).   I think we might need a set of runtime switches and API to let the user set the mapping, similar to numactl.  Defaulting to above probably makes sense, as that is a relatively efficient use of CPUs.

commit f4038e36 (2022-07-12): Move thread pool and execution profiler into the context. (#3477)

Fixes #3454