verilator/verilator #3458: Parallelization of Verilator code

mglb, 2022-05-31:
Hi!

I'm looking at the possibility of making (parts of) Verilator code parallel in order to speed up the verilation process.
Which stages are reasonably good candidates for parallelization? What would need to be reimplemented?

So far I've found that `V3EmitC::emitcImp()` looks promising. Excluding file writing (i.e. `fputc`) itself, it takes about 10% (17s) of the whole verilation time in one bigger project I'm testing.
Looks like the stage handles independent(?) subtrees in a loop. The main thing preventing parallelization is the use of node's `user` properties, which are currently locked globally.

gezalore, 2022-05-31:
You are on the right track. Generally passes that operate on trees independently could be run in parallel. There are many of these, though all the global state, especially the generation numbers and in-use-flags in the AstNode::user* are somewhat obtrusive. I don't have a magic bullet here, other than review each pass if it might be safe or can be made safe easily. The `--stats` file lists how long each pass takes if you want to look for candidates. Unfortunately the really long running ones (e.g.: multi-threaded partitioning and variable ordering) tend to be hard to parallelize, so there might be only a limited amount of low hanging fruit. We also need to be careful about doing this in a way that is maintainable and testable to avoid hard to debug issues. Maybe @wsnyder has some thoughts.

FWIW I have some WIP changes from a long time ago that did parallel emit (and gained about 10% speed as you suggest), so that part is definitely doable, feel free to nick any of it: https://github.com/gezalore/verilator/commit/5eac78fca10241fbcc49f5f7ca5032a0997e5282

wsnyder, 2024-09-08:
Closing, as we've been adding parallel code as pulls are accepted.

closed 2024-09-08 by wsnyder