verilator/verilator #3464: XPM Primitives on Verilator

TomO700, 2022-06-12:
Hi,

I am trying to simulate a design containing Xilinx XPM Primitives using Verilator and Cocotb. I get the error below, which I believe is due to the XPM. Is there a way/switch to simulate designs containing XPM using verilator?

%Error: top.sv:50:5: Cannot find file containing module: 'xpm_memory_sdpram'

wsnyder, 2022-06-12:
That specific message means you aren't giving the verilator command line the file that has xpm_memory_sdparam or there isn't a search path with xpm_memory_sdparam.v in it.

Note Verilator doesn't support primitives (#468) but you'd get a clearer error if that was what you were hitting, and I'm not sure you mean "primitive" in the Verilog sense in your question, as the name of the file suggests parameters not gate primitives.

closed 2022-06-12 by wsnyder

TomO700, 2022-06-12:
_**xpm_memory_sdparam**_ is a XPM (Xilinx Parameterized Macros)
https://docs.xilinx.com/r/en-US/ug974-vivado-ultrascale-libraries/Xilinx-Parameterized-Macros
error is exactly what I have given
_%Error: top.sv:50:5: Cannot find file containing module: '_**xpm_memory_sdpram**_'_

This is not a file I hand coded. Thus the question on how to use it. 
Will you be able to help? Thanks

wsnyder, 2022-06-12:
You aren't giving the verilator command line the file that has xpm_memory_sdparam or there isn't a search path with xpm_memory_sdparam.v in it.  If you don't know what file you need, do a grep of the source code to see what file has it.

TomO700, 2022-06-12:
Thanks. I found the file and gave it. However the following error pops up. I believe its for the verilog '**deassign**'
_%Error-UNSUPPORTED: /mnt/D/d/OB_tb/xpm_memory.sv:5385:12: Unsupported: Verilog 1995 deassign
 5385 |            deassign gen_rd_a.douta_reg;_
 
 Will this have to be replaced?

wsnyder, 2022-06-12:
Yes, deassign is not supported.

TomO700, 2022-06-12:
Thanks. Any idea what the alternative is?

wsnyder, 2022-06-12:
Depends on the code (do not post it due to copyright).  Often a force/release or adding some temporary signal to prevent the deassign would work.  Basically rewrite it as if you were synthesizing the code.