verilator/verilator #3470: Wrong expression evaluation  results

algrobman, 2022-06-16:
While simulating swerv EL2 core (https://github.com/chipsalliance/Cores-SweRV-EL2) and running coreMark test I've noticed that the resulting score differs when running  verilator 4.222 vs irun.

Verilator 4.222 2022-05-02 rev v4.222-24-g4618fd5a

Verilator 4.210 yields  the same score as irun.

run command is:
make -f $RV_ROOT/tools/Makefile verilator TEST=cmark_iccm GCC_PREFIX=kuku

 results:
 
 **Iterat/Sec/MHz   : 2.61**
Iterations       : 1
Compiler version : GCC9.2.0
Compiler flags   : -g -O3 -funroll-all-loops

when running with irun:
make -f $RV_ROOT/tools/Makefile irun TEST=cmark_iccm GCC_PREFIX=kuku

**Iterat/Sec/MHz   : 3.37**
Iterations       : 1
Compiler version : GCC9.2.0
Compiler flags   : -g -O3 -funroll-all-loops

I've traced this to wrong evaluation of ECC decoder in LSU unit rvecc_decode module from design/lib/beh_lib.sv  asserting single bit error signal.
this module is instantiated in
design/lsu/el2_lsu_ecc.sv file 
under lsu_ecc_decode_lo name . single_ecc_error_lo_any signal should not be asserted in this simulation !

ECC decoder is pure combinatorial module ..

wsnyder, 2022-06-16:
Can you please make this into a self-testing self-contained minimal test case?  I'd also suggest try -Oo.

algrobman, 2022-06-16:
it works fine with -Oo,

Not sure how to write a test and get results from combo module - example,please.

wsnyder, 2022-06-16:
Look at test_regress/t/t_EXAMPLE.v.  This drives a random pattern in, then you use icarus to determine the right return value.

algrobman, 2022-06-17:
I know how to write TB for such things for Xcellium or VCS, but not sure how to run it on verilator ...

algrobman, 2022-06-18:
Tried to create a small example, (driving ecc_decoder module inputs from recorded file) similar to #3197, but it simulates correctly. More logic should be exposed probably ..

algrobman, 2022-06-18:
the mismatch happens 1st time near clock 64 of the simulation. :

```
        63 :      #21 0 ee006f38 f00403b7  t2=f0040000 ; lui     t2,0xf0040000
        64 :      #22 0 ee006f3c 0ec72a83  s5=00000001 ; lw      s5,236(a4) [f00400ec]
```
when din = 1 ...

sdiff exec.log exec.log_verilator

```
        63 :      #21 0 ee006f38 f00403b7  t2=f0040000 ; lui            63 :      #21 0 ee006f38 f00403b7  t2=f0040000 ; lui 
        64 :      #22 0 ee006f3c 0ec72a83  s5=00000001 ; lw             64 :      #22 0 ee006f3c 0ec72a83  s5=00000001 ; lw  
        65 :      #23 0 ee006f40 0000cd22              ; c.sw |         68 :      #23 0 ee006f40 0000cd22              ; c.sw
        66 :      #24 0 ee006f42 0000cb26              ; c.sw |         69 :      #24 0 ee006f42 0000cb26              ; c.sw

```
the module scope is : tb_top.rvtop.swerv.lsu.ecc.Gen_dccm_enable.lsu_ecc_decode_lo

algrobman, 2022-06-18:
Can somebody take a look on resulted C function with optimization and with -Oo switch to see what can be broken in this module?  ( tb_top.rvtop.swerv.lsu.ecc.Gen_dccm_enable.lsu_ecc_decode_lo)

yTakatsukasa, 2022-06-20:
Hopefully I will take a look at it this weekend. I need to start from learning how to run coremark on the design.

algrobman, 2022-06-24:
any positive news for me?

algrobman, 2022-06-27:
@yTakatsukasa thanks for looking in that ..

yTakatsukasa, 2022-06-28:
I guess simply because the triggering code does not appear often.

From AST, the input looks as below (note that intermediate wires/modules are omit).
```systemverilog
(^dccm_rd_data_lo[31:0]) ^ (^dccm_rd_data_lo[38:32])
```

In most cases, people would write as below if possible
```systemverilog
^dccm_rd_data_lo
```

algrobman, 2022-06-29:
Actually I was trying to simulate only ecc_decoder block , which does include offending code ..

algrobman, 2022-06-29:
(here is my example)
[ecc.tar.gz](https://github.com/verilator/verilator/files/9006644/ecc.tar.gz)
 it does not fail...
BTW original RTL has this code:

  assign ecc_check[6] = ((^din[31:0])^(^ecc_in[6:0])) & ~sed_ded;

where din and ecc_in were inputs of the  module . 

Looks like verilator inlines the module after CPU compilation but does not do this in my small example ..

algrobman, 2022-06-29:
Any date projections for new release?

commit 9f37cef1 (2022-07-05): Fix #3470 of incorrect bit op tree optimization (#3476)

* Tests: Add a test to reproduce #3470

* Update LSB during return path of traversal. No functional change is intended.

* Introduce LeafInfo::m_msb

* Update LeafInfo::m_msb when visitin AstCCast

* Internals: Add comment, reorder. No functional change is intended.

* Delete explicit from copy constructor to fix build error.

* Update Changes

* Internals: Remove unused parameter. No functional change is intended.

* Tests: Add explanation to t_const_opt.
closed 2022-07-05 by yTakatsukasa