verilator/verilator #3472: example_cc in the documentation failing to compile

aproxp, 2022-06-20:
The example given in the https://github.com/verilator/verilator/blob/master/docs/guide/example_cc.rst fails to compile with the given command `verilator -Wall --cc --exe --build sim_main.cpp our.v`, with the following compiler error:
```
../sim_main.cpp:4:7: error: ‘VerilatedContext’ was not declared in this scope; did you mean ‘VerilatedMutex’?
```
This is with `Verilator 4.106 2020-12-02 rev v4.106`, which I am just starting to play around with and later want to use with cocotb.
Anyway, since the examples given in https://github.com/verilator/verilator/tree/master/examples work just fine, I could just rewrite this documentation page `make_hello_c` example.
Is that alright for me to do? If so, since I did not contributed to the project in the past, do I just edit the page in question and submit a pull request?

Cheers.

wsnyder, 2022-06-20:
First, thanks for trying the examples and getting started!

Please note the examples in github are for running against the verilator version in github, that is 4.225.   I suspect when you say "the examples in tree/master/examples work fine" you are running those examples against the sources in github, that is 4.225, not 4.106.  VerilatedContext was indeed introduced with 4.200.

To use the older version you'll need to look at the documentation for that version (which presumably came along with however you got 4.106), or better, update verilator to the most recent version.  Specifically I would suggest always building locally (or use latest stable from docker) instead of any RPM that comes with your OS.

I just confirmed docs/guide/example_cc.rst does run against the sources there.

Closing as answered, feel free to follow up.

closed 2022-06-20 by wsnyder

aproxp, 2022-06-20:
Thank you for clearing that up. I compiled the sources and installed from the 4.106 tag, which is also the version of the examples that I ran against. And as you said, I took the first example from the documentation at master, which caused all of this.
For anybody who wants to use the (recommended for now) 4.106 with cocotb and stumbles upon this thread, you can get the documentation for 4.106 by checking out that version tag and running `make docs` (depending on `libpod-latex-perl texlive-latex-recommended texlive-latex-extra`)

andymancan2, 2023-02-17:
I there a way to install 4.2+ with "Package Manager Quick Install"?
It appears I am running:  Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad

wsnyder, 2023-02-17:
@andymancan2 for new questions please post a new issue.  But to answer, as to the latest version you can get with your distribution, that is a question for your distribution's package maintainers.  Generally they are always behind, so better to install from sources.