verilator/verilator #3474: always_comb does not warn about latch

pinkcatfly, 2022-06-27:
Version : Verilator 4.220 2022-03-12 rev UNKNOWN.REV

Used on MACOS installed by brew

```
module test ();
    logic bit_0;
    logic bit_1;
    logic bit_2;
    logic bit_3;
    logic [7:0] vector_0;

    always_comb begin
        if (bit_0) begin
            bit_1 = 1;
        end
    end
    
    always_comb begin
        unique case (vector_0)
            8'b00000001: begin
                if (bit_0) begin
                    bit_2 = 1;
                end
            end
            default: begin
                bit_2 = 0;
            end
        endcase
    end

    always_comb begin
        unique case (vector_0)
            8'b00000001: begin
                if (bit_0) begin
                    bit_3 = 1;
                end else begin
                    bit_3 = 0;
                end
            end
            default: begin
                bit_3 = 0;
            end
        endcase
    end
endmodule
'''

The only warning I get is (except for signal not driven or used)

`Latch inferred for signal 'test.bit_1' (not all control paths of combinational always assign a value)`

However bit_2 in the first always_comb block is also a latch, I suppose.

MG525, 2022-08-13:
@wsnyder 
I investigated the issue. However, the latch is detected  correctly v3warn in latchCheck is not printed and by replacing v3warn by v3error it is working fine.

There might be a blocker for duplicated warnings of the same code?

MG525, 2022-08-13:
Replacing "LATCH" error code with "NOLATCH" or "INSECURE" is also working fine!

something is special with "LATCH" error code that needs to be investigated

MG525, 2022-08-17:
So I debugged a little and the issue was warning for latch is turned off when a case statement is hit within combinational always

**V3Case.cpp**
`if (m_alwaysp) m_alwaysp->fileline()->warnOff(V3ErrorCode::LATCH, true);`

so I can see with the current infrastructure for latch detect that CASE is not supported and if the warning is turned on then we can get FALSE Negative of FALSE Positive situations for latch detection within CASE

I think the next steps for this might be the support of CASE in latch detect? I can see through Test cases comments that this was postponed due to the complex situations of CASEX and CASEZ so maybe we can start with CASE and then extend later?

What do you think @wsnyder ?

wsnyder, 2022-08-17:
I'm not sure casex/z is that much harder as for any case statement the hard part is to check for completeness, but making any improvement in the handled checks seems an improvement.

margej, 2023-06-08:
The latch warning is only disabled for case statements which have not already been identified as complete (I.e. those where isCaseTreeFast(nodep) == false)

The reasoning behind this stems from the following comment:
https://github.com/verilator/verilator/issues/1609#issuecomment-749682092

As far as I could tell at the time, this was non-trivial to solve (at least for me!)