verilator/verilator #3478: Incorrect logic optimization

ngraybeal, 2022-06-28:
Failure seen on version 4.222

Ran on 4.224 and issue was resolved. I glanced at the release notes for 4.224 and nothing stood out to me as a fix for this, so raising this issue just in case.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```
module WrapTest(
  input   clock,
  input   reset,
  output  io_success
);
  reg [2:0] small_;
  wire [3:0] nextSmall = small_ + 3'h1;
  reg [4:0] largeOpt;
  wire [4:0] nextLargeOpt = largeOpt + 5'h1;
  wire [4:0] _T_10 = ~nextLargeOpt;
  wire [4:0] _T_11 = largeOpt & _T_10;
  assign io_success = nextSmall[3] & _T_11[4];
  wire [3:0] _GEN_5 = reset ? 4'h0 : nextSmall;
  always @(posedge clock) begin
    small_ <= _GEN_5[2:0];
    if (reset) begin
      largeOpt <= 5'h0;
    end else if (nextSmall[3]) begin
      largeOpt <= nextLargeOpt;
    end
  end
endmodule
```

io_success ends up with an incorrect optimization. the last "bitwise and" should be shifted as well. 
```
    vlSelf->io_success = (IData)(((((IData)(vlSelf->WrapTest__DOT__largeOpt) 
                                    >> 4U) & (((IData)(1U) 
                                               + (IData)(vlSelf->WrapTest__DOT__small_)) 
                                              >> 3U)) 
                                  & (~ ((IData)(1U) 
                                        + (IData)(vlSelf->WrapTest__DOT__largeOpt)))));
```

What OS and distribution are you using?
rhel8

yTakatsukasa, 2022-06-28:
Thank you for the report.
It looks this issue is same as #3445.
It is fixed by #3453 & #3459 that is included in v4.224, but not in v4.222.

My bad for not updating Changelog

ngraybeal, 2022-06-28:
Oh perfect. Just wanted to make sure it was known. Thanks.

closed 2022-06-28 by ngraybeal