verilator/verilator #3481: coverage/trace

quantrpeter, 2022-07-03:
Hi all
   1. —coverage-line only give me a report about which line of code has run in total, but i need to track which one line of code is executed when clock change, any way i can do this?
   2. can I self define a macro, called in verilog to c? something like $display
   3. if i want to manually insert some code behind each assign statement, i should write my own verilog preprocessor or using verilator’s visitor?
thanks

wsnyder, 2022-07-03:
2. Yes, make an e.g. DPI import function and call it.
3. Yes, you could do that too, using e.g. Verilog-Perl as a standalone editor, or make a Verilator pipe filter to process the code on-the-fly.

If you have more questions please post followups, don't make a new issue, thanks.

closed 2022-07-03 by wsnyder

quantrpeter, 2022-07-04:
thanks Mr @wsnyder