verilator/verilator #3484: $display and $write with empty argument breaks directly following $display or $write

Grulfen, 2022-07-05:
Verilator 4.224 2022-06-19 rev v4.224

Example .sv:
```verilog
module tjo();

initial begin
  $display("testing", "");
  $display("this will not be printed");
  $finish();
end

endmodule
```

This will generate the following code, note the `\0`s in call to `VL_WRITEF`:
```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Body
    VL_WRITEF("testing\000\000\000\000\000\000\000\000\nthis will not be printed\n");
    VL_FINISH_MT("tjo.sv", 6, "");
}
```

As far as I can se `VL_WRITEF` is just a `printf` by default. The null terminators will cause this code to only print:
```
testing
```

If the empty argument is removed:

```verilog
module tjo();

initial begin
  $display("testing");
  $display("this will be printed");
  $finish();
end

endmodule
```

The generated code will look like this:
```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Body
    VL_WRITEF("testing\nthis will be printed\n");
    VL_FINISH_MT("tjo.sv", 6, "");
}
```

and the code will print 
```
testing
this code will be printed
```

I also noticed that I dont see this issue if the first argument to `$display` is an empty string:

```verilog
module tjo();

initial begin
  $display("", "testing");
  $display("this will be printed");
  $finish();
end

endmodule
```

```cpp
VL_ATTR_COLD void Vtjo___024root___initial__TOP__0(Vtjo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtjo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtjo___024root___initial__TOP__0\n"); );
    // Body
    VL_WRITEF("testing\nthis will be printed\n");
    VL_FINISH_MT("tjo.sv", 6, "");
}
```

sbmpost, 2022-07-08:
Maybe related to #3488 ?

wsnyder, 2022-07-09:
Thanks for the good examples to fix.

closed 2022-07-09 by wsnyder