verilator/verilator #3485: Verilator VCD dump with signals containing colon is rejected by GTKWave

C-Elegans, 2022-07-06:
When Verilating a module that contains wires with a colon in their names, the VCD produced by verilator is rejected by GTKWave. 

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```module test_mod(b, clk, a);
    input clk;
    input a;
    output b;
   
    wire \test_sig_with_colon:3419 ;

endmodule
```
I wasn't aware that colons were allowed in verilog identifiers, however yosys produces verilog with them in it and verilator accepts it as valid. 

What 'verilator --version' are you using?  Did you try it with the git master version?
I am running verilator built from commit 9f37cef1bb.

What OS and distribution are you using?
Ubuntu 20.04

May we assist you in trying to fix this in Verilator yourself?

Yes please. I have written a patch that fixes this locally, but I'm not entirely sure if this should be fixed in verilator or GTKWave, or whether this is the right way to fix it

wsnyder, 2022-07-06:
Does the vcd file properly have the backslash and colon?

If so, this is a gtkwave viewer bug.

If there is a colon but no leading backslash it is a verilator bug, and please fix to add the backslash as it does with other characters.

wsnyder, 2022-07-09:
The : properly comes out in the VCD, so this is a GTKwave bug.

closed 2022-07-09 by wsnyder