verilator/verilator #3491: missing ';' when convert single line systemverilog queue operation into c++

qian-gu, 2022-07-12:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```systemverilog
typedef struct packed {
  logic [31:0]  araddr;
  logic [7 : 0]  arid;
  logic [7 : 0]  arlen;
} ar_t;

ar_t  ar;
ar_t  ar_queue[$];

// only one single line queue operation in if block
always @(negedge clk) begin
  if (push_en) begin
    ar_queue.push_back(ar);
  end
  if (pop_en) begin
    void'(ar_queue.pop_fornt());
  end
end
```

in the previous example, the converted result is like,

```c++
if (balabala) {
   vlSelf->tb_top__DOT__imem__DOT__ar_queue.pop_fornt()}
```

and g++ says, 

> Vtb_top__024root_DepSet_h9ad02603__44.cpp:2106:61: error: expecetd ';' before '}' token
>        vlSelf->tb_top__DOT__imem__DOT__ar_queue.pop_fornt()}

after adding a redundant line(`tmp <= 0`), the compilation and simulation finish successfully.

```systemverilog

// add redundant code to ensure there are multiple lines in if block
logic tmp;
always @(negedge clk) begin
  if (push_en) begin
    ar_queue.push_back(ar);
  end
  if (pop_en) begin
    tmp <= 0;
    void'(ar_queue.pop_fornt());
  end
end
```

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 4.222 2022-05-02` compiled from source code on local machine.

What OS and distribution are you using?

```
Red Hat Enterprise Linux Server
Release 6.5 (Santiago)
```

wsnyder, 2022-07-12:
Similar to #3364.

drewranck, 2022-08-11:
Sorry I lost track of my old PR. If you need a fix, I'll be starting a new PR from:
https://github.com/drewranck/verilator/tree/issue-3364-try3

qian-gu, 2022-08-12:
Thanks for your kindness, I'd like to make a test after the PR.

drewranck, 2022-08-12:
Looks like the PR is in master now, so go for it!

If you find something you don't like or a missing test, you can add it to this:
https://github.com/verilator/verilator/blob/master/test_regress/t/t_void_queue_ops.v#L156

qian-gu, 2022-08-15:
Great job, the problem is solved.

closed 2022-08-15 by qian-gu