verilator/verilator #3495: --output-split-cfuncs segfaulting or generating bad code with a $dump systemtask

VarunKoyyalagunta, 2022-07-13:
I've added two tests here -
https://github.com/verilator/verilator/compare/master...VarunKoyyalagunta:verilator:bug/trace-split-cfuncs

With a `$dump` systemtask and cfuncs being split, verilator segfaults because the file pointer is null.

If there is also a DPI export, the output `_traceDump` code is written into the middle of the DPI code, and gcc fails.

The tests have `--output-split-cfuncs 1` to simplify them, but in our codebase we are seeing the same errors without `--output-split-cfuncs` specified.

Tried with 4.220 and latest master as of writing (f4efcbde5c1a5f700a7db4cfb4e8ffe5e19bf192)

I have a naive fix here that causes my tests to pass -
https://github.com/VarunKoyyalagunta/verilator/compare/bug/trace-split-cfuncs...VarunKoyyalagunta:verilator:bugfix/trace-split-cfuncs-write-to-base

wsnyder, 2022-07-14:
Thanks for proposing a fix. I think a better fix is the m_fpBase's near the one you added should be removed.

Might you be able try that and convert this into a pull request here?

VarunKoyyalagunta, 2022-07-14:
Hi @wsnyder, I'm not sure if I'm understanding your suggestion, but this did not work for me -
https://github.com/VarunKoyyalagunta/verilator/compare/bug/trace-split-cfuncs...VarunKoyyalagunta:verilator:bugfix/failed-trace-split-cfuncs-no-write-to-base

The problem, as I understand it, is this `closeSplit()` is invalidating `m_ofp`
https://github.com/VarunKoyyalagunta/verilator/blob/768fef005a6aa65c709bdd54b9a4185529d7c9ed/src/V3EmitCSyms.cpp#L815

I'm not really sure the `traceDump` code really belongs in this part of the function. Everything after constructor generation starts seems to assume it's the final function in the file. I created a pull request where I move up the traceDump generation to before the constructor is generated, and that works - https://github.com/verilator/verilator/pull/3497

wsnyder, 2022-08-25:
Thanks for your pull!

closed 2022-08-25 by wsnyder