verilator/verilator #3496: Potential bugs detected using clang-tidy.

phelter, 2022-07-14:
Verilator version 4.222.
OS Distribution(s): Ubuntu 22.04 and Ubuntu 18.04
C/C++ Compiler: Clang-13 and Clang-14

I've been using verilator for unit and system level tests with googletest as the driver test framework and it is working quite well.  To ensure I have the best quality code for both the drivers to the verilated testbenches and to confirm the quality of the verilator code I am using, I ran the code through several linters (CppLint, CppCheck, Clang-Tidy).

The base code passes both CppLint and CppCheck no issues; however, clang and clang-tidy both show quite a few things that are potential bugs.  There are also a lot of `coding style preferences` checks in clang-tidy but most of those are widely accepted and make the code easier to read.

I'm using cmake for compiling things - Note the clang-compilation is with overall compile flags:
```cmake
add_compile_options(
    $<$<COMPILE_LANG_AND_ID:CXX,GNU>:-fdiagnostics-color=always>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-fcolor-diagnostics>

    $<$<COMPILE_LANG_AND_ID:CXX,Clang,GNU>:-Wall>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang,GNU>:-Wextra>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Weffc++> # Note GNU too many errors in verilator code.
    $<$<COMPILE_LANG_AND_ID:CXX,Clang,GNU>:-Werror>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wpedantic> # Note not used in GNU to allow C99 designated initializers.
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Weverything>

    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wno-c++98-compat>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wno-c++98-compat-pedantic>
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wno-c++20-designator> # Allow C++20 designators for all c++ versions.

    # Allow clang and gnu pramgas.
    $<$<COMPILE_LANG_AND_ID:CXX,GNU>:-Wno-unknown-pragmas>

    # Documentation types
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-fcomment-block-commands=retval> # Doesn't recongize doxygen retval
    $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-fcomment-block-commands=copydetails> # Doesn't recongize doxygen copydetails
)
```

I've added the portion of the CMakeLists.txt file that compiles the common/base code that is added to all other testbenches below.  In this cmake script the items under `target_compile_options` and my own function `target_clang_tidy_definitions` that removes those specific checks/issues (i.e. opt-out).

The most notable ones are - that have the highest potential for bugs/issues.
```cmake
        $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wno-thread-safety-analysis>
        $<$<COMPILE_LANG_AND_ID:CXX,Clang>:-Wno-conditional-uninitialized>

        -clang-analyzer-core.CallAndMessage
        -clang-analyzer-core.NonNullParamChecker
        -clang-analyzer-core.NullDereference
        -clang-analyzer-core.UndefinedBinaryOperatorResult
        -clang-analyzer-optin.cplusplus.UninitializedObject
        -cppcoreguidelines-avoid-non-const-global-variables
        -cppcoreguidelines-c-copy-assignment-signature
        -cppcoreguidelines-init-variables
        -cppcoreguidelines-virtual-class-destructor
        -performance-noexcept-move-constructor
```

Here's the snippet of the cmake.
```cmake
    find_package(verilator 4.222 REQUIRED
        HINTS
        /usr/local
        $ENV{VERILATOR_ROOT}
        ${VERILATOR_ROOT}
    )

    if (NOT verilator_FOUND)
        message(FATAL_ERROR "Verilator was not found. Either install it, or set the VERILATOR_ROOT environment variable")
        # Install instructions here: https://verilator.org/guide/latest/install.html
    endif()

    option(VERILATOR_ENABLE_SC "Enable System C" OFF)
    option(VERILATOR_THREADS   "Number of Threads"  0)
    if (NOT TARGET Verilator::base)
[... 144 lines cut ...]
              -hicpp-explicit-conversions
              -hicpp-member-init
              -hicpp-noexcept-move
              -llvm-include-order
              -misc-unconventional-assign-operator
              -modernize-use-nodiscard
              -performance-noexcept-move-constructor
              -readability-avoid-const-params-in-decls
              -readability-identifier-length
              -readability-implicit-bool-conversion
              -readability-make-member-function-const
              -readability-redundant-access-specifiers
        )
  
        target_link_libraries( verilator_base
            PUBLIC
            atomic # For some reason missing __atomic_is_lock_free definition.
        )
        if( VERILATOR_ENABLE_SC )
            verilator_link_systemc(verilator_base)
        endif()
    endif()
```

I am willing to help out on Verilator, however I don't work on opensource projects below C++17 (preferably C++20) and first ensure that the code is to a particular standard, eg. most of the above are cleaned up before proceeding on any work, which I'm willing to help out on (first step is identification).

wsnyder, 2022-07-14:
We'd love to have help.  I'm hoping C++17 isn't an absolute requirement for your help, as many industrial users of Veriator are stuck on older compilers so we must stay about 8 years behind.

The github actions mostly run with GNU Make versus GCC, if you set the VERILATOR_AUTHOR_SITE environment variable or configure appropriately (see https://github.com/verilator/verilator/blob/master/docs/internals.rst#id22) the checks that are clean will run.

There is also a "make clang-tidy" target that in theory was once clean.  This isn't part of the regressions, but should be, and would welcome an appropriate addition to the github actions.  I'd prefer to get such a check into the regressions before enabling new rules to check.

I'd certainly welcome reasonable code cleanups and new rules to be enabled.

phelter, 2022-07-17:
It is unfortunate that you have the requirement to stay on older compilers.
Would it be possible to consider the next major version of Verilator (5.x) to no longer support older compiler versions and allow eg. version 4.x to be available for industrial users stuck on those older compilers.  If that version is working for those individuals it means they most likely won't migrate to newer versions of Verilator anyways due to the fact that they are hesitant on changing their OS or compiler version.

Is there any chance of migrating to `cmake + ninja-build` from `autocfg + make` as well (first supporting both, and then hopefully using just one)?  I find that this has become the industry standard for C/C++ projects and allows for a significant improvement in cross-platform support over the existing make structure that you have in place.  It also allows for a simple way of producing *.deb or *.rpm for making installation much simpler.  I do have `make` experience, however to create the same environment I have for my projects in `make` vs. `cmake and ninja-build` it would be a significant undertaking migrating that back to `make`.

If those are options you are open to, I'm happy to look into that undertaking and work to adding more checks. Including dynamic analysis using Sanitizers.

phelter, 2022-07-20:
Ok - if it's a hard requirement that's quite unfortunate.  With the use of docker, an OS managed by IT for it's servers is still achievable, but also allows DevOp(s) or Designers/Verifiers of a given project to decide what versions of installed compilers and system tools to have on a given docker image.  This makes most of these types of concerns irrelevant.  The one thing you require from IT is to allow and support docker on those OS'es.  Agree with you about not tying a verification tool to an OS, I only only suggested increasing the minimum requirements of build tools to support GCC-7.5 (2017) or higher and/or Clang-13+.  Both of which are capable of being installed on older versions of Ubuntu/CentOS/RHEL.

Yes there is a learning curve for cmake, but once setup and configured correctly it can make the build process much easier.  How about I take a stab at getting my stand-alone cmake build environment providing all of the same build capabilities that you currently have with the existing tools and then see where that leads?

wsnyder, 2022-10-15:
First, it's agreed that the make process doesn't have good documentation. As always we appreciate pull requests to documentation, if you have something in mind.

>How do I build the output of verilator --cc (without also using --exe)

It sounds like you want your own top level Makefile so you can control it, this then needs to link three things, your main() object, the Verilated design library archive, and the common verilated library files.

The default generated files by default make an archive library of the verilated design.

For the common Verilated library this would be easier if there was a libverilated.a, but there is not yet, but this is in the plans as described in #3507.  So for now you need to use the list of global files that Verilator generates for you in VM_GLOBAL_FAST/VM_GLOBAL_SLOW.

Anyhow, here is how to do with a Makefile without --exe, or passing any .cpp files, or using verilated.mk.

```
# Write this into my.mk
# Note VM_ variables are set in __classes.mk
include Vtop_classes.mk

VPATH = $(VERILATOR_ROOT)/include ..

CPPFLAGS += -I$(VERILATOR_ROOT)/include -I.

MY_GLOBAL_OBJS = $(addsuffix .o,$(VM_GLOBAL_FAST) $(VM_GLOBAL_SLOW))

simx: $(MY_GLOBAL_OBJS) sim_main.o Vtop__ALL.a
        $(CXX) $(LDFLAGS) $^ $(LOADLIBES) $(LDLIBS) $(LIBS) $(SC_LIBS) -o $@

sim_main.o: sim_main.cpp

%.o: %.cpp
        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) -c -o $@ $<
```

Run this:

```
# Make sure your VERILATOR_ROOT is set before trying this
mkdir t
cd t
cp $VERILATOR_ROOT/examples/make_tracing_c/{*.cpp,*.v} .
$VERILATOR_ROOT/bin/verilator --cc --trace top.v
make -C obj_dir -f Vtop.mk  # build obj_dir/Vtop__ALL.a
make -C obj_dir -f ../my.mk  # build obj_dir/sim_main and simx
obj_dir/simx
```

>VPATH

VPATH avoids needing directory paths in front of each source file in make. IMO this is more readable when there are multiple directories with sources; it would have been a mess to write the above without it.

Note also absolute paths must be avoided to allow ccache to work well when there are many users using different directories.

>more generally (2) why is verilator generating makefiles anyway?  ... [Verilator should not be] taking over the build process from the user.

Long ago it was closer to as you describe.  Over time Verilator has become more like other simulators, wherein is you run it and it deals with all of the details and you get a simulation binary out.  The --binary option is the latest in this trend.  This is now I suspect the more common use case; many users don't use C++ or make, and don't want to.  So, in short, we won't be removing the option of building under Verilator.

We still embrace those power users that need to be masters of their own build environments, and will continue to support building outside of Verilator.

>Might I suggest a reorganization to make the generated makefile more "low powered", that is, just use fewer features of make?

History has pushed it into what it does now for better or worse.  Part of the reason for the current structure is that:

1. It mostly separates project-specific variables/rules from general common rules.

2. It allows users to write their own Makefiles from scratch, using just the file list in {design}_classes.mk (as above).

3. It allows multiple designs or libraries to be built in parallel using the same common rules in include/verilated.mk.

4. Changing it may break things, and for those that have complicated enviornments change can be difficult.

That said, improvements are welcome.

wsnyder, 2022-10-15:
Above was posted to #3496 by mistake, I'm deleting it.

wsnyder, 2022-11-19:
Think the points here were addressed, but feel free to follow up.  I don't think I'm ready to consider a cmake conversion at this time.

closed 2022-11-19 by wsnyder