verilator/verilator #3499: What is the point of V3Coverage?

burntfalafel, 2022-07-16:
Title says it all - I am confused on why anyone would use this source file in the first place. From what I have seen running the coverage command in the [_make_tracing_c_](https://github.com/verilator/verilator/tree/master/examples/make_tracing_c) example, at a high verbosity `--debugi 5`, I get the following output: 
```
-- VERILATE ----------------
/home/ginjama/workspace/work/verilator/bin/verilator -cc --exe -x-assign fast -Wall --trace --assert --coverage -f input.vc top.v sim_main.cpp
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c0e460 <e773#> {d14ax} @dt=0x5564f3c19fc0@(G/w1)  clk INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c0e980 <e781#> {d15ax} @dt=0x5564f3c19fc0@(G/w1)  reset_l INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c13ce0 <e789#> {d17ax} @dt=0x5564f3c0f090@(G/w2)  out_small OUTPUT [VSTATIC]  WIRE
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c14810 <e797#> {d18ax} @dt=0x5564f3c14150@(G/w40)  out_quad OUTPUT [VSTATIC]  WIRE
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c18520 <e805#> {d19ax} @dt=0x5564f3c14c80@(G/w70)  out_wide OUTPUT [VSTATIC]  WIRE
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c18f10 <e929#> {d20ax} @dt=0x5564f3c0f090@(G/w2)  in_small INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c19950 <e930#> {d21ax} @dt=0x5564f3c14150@(G/w40)  in_quad INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c15f80 <e931#> {d22ax} @dt=0x5564f3c14c80@(G/w70)  in_wide INPUT [VSTATIC]  PORT
- V3Coverage.cpp:385:  IF: IF 0x5564f3c07f80 <e351#> {d38ah}
- V3Coverage.cpp:422:    COVER-branch: IF 0x5564f3c07f80 <e351#> {d38ah}
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c145d0 <e1012#> {d14ax} @dt=0x5564f3c19fc0@(G/w1)  __Vtogcov__clk MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c18cd0 <e1031#> {d15ax} @dt=0x5564f3c19fc0@(G/w1)  __Vtogcov__reset_l MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c1a090 <e1050#> {d17ax} @dt=0x5564f3c0f090@(G/w2)  __Vtogcov__out_small MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c201e0 <e1181#> {d18ax} @dt=0x5564f3c14150@(G/w40)  __Vtogcov__out_quad MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c3d740 <e3592#> {d19ax} @dt=0x5564f3c14c80@(G/w70)  __Vtogcov__out_wide MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c6d5d0 <e7803#> {d20ax} @dt=0x5564f3c0f090@(G/w2)  __Vtogcov__in_small MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c6ed50 <e7934#> {d21ax} @dt=0x5564f3c14150@(G/w40)  __Vtogcov__in_quad MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3c8a930 <e10345#> {d22ax} @dt=0x5564f3c14c80@(G/w70)  __Vtogcov__in_wide MODULETEMP
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c21620 <e713#> {e10ak} @dt=0x5564f3c19fc0@(G/w1)  clk INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c21af0 <e721#> {e11ak} @dt=0x5564f3c19fc0@(G/w1)  reset_l INPUT [VSTATIC]  PORT
- V3Coverage.cpp:268:     Toggle: VAR 0x5564f3c228c0 <e950#> {e15ap} @dt=0x5564f3c07d70@(G/w32)  count_c [VSTATIC]  VAR
- V3Coverage.cpp:385:  IF: IF 0x5564f3c25ab0 <e473#> {e17ah}
- V3Coverage.cpp:385:  IF: IF 0x5564f3c259a0 <e472#> {e25ak}
- V3Coverage.cpp:422:    COVER-branch: IF 0x5564f3c259a0 <e472#> {e25ak}
- V3Coverage.cpp:422:    COVER-branch: IF 0x5564f3c25ab0 <e473#> {e17ah}
- V3Coverage.cpp:481:  COVER: COVER 0x5564f3c27b40 <e527#> {e40ae}
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3cbb720 <e14567#> {e10ak} @dt=0x5564f3c19fc0@(G/w1)  __Vtogcov__clk MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3cbbb50 <e14586#> {e11ak} @dt=0x5564f3c19fc0@(G/w1)  __Vtogcov__reset_l MODULETEMP
- V3Coverage.cpp:266:     Disable Toggle: Not relevant signal type VAR 0x5564f3cbc1a0 <e14605#> {e15ap} @dt=0x5564f3c07d70@(G/w32)  __Vtogcov__count_c MODULETEMP

```

And I am looking for clarifications for the following things:
- What does the address `0x5564f3c0e460` mean? 
- Toggling on and off - what does it signify? From what I have understood [from the code](https://github.com/verilator/verilator/blob/b0e796ca83cd05a6f72674c89e9af1eaa9c71366/src/V3Coverage.cpp#L90), it signifies verilator won't trace the code which has a leading underscore, or exceeds a certain max-width. Then why are wires from the `sub.v` being toggled on and off?
- And finally the miscellanous angle brackets and curly bracket infromation. Is this even needed? 

Thanks for any of the replies :)!

wsnyder, 2022-07-16:
I'm confused by your comment "Is this even needed", debug dumps are for debugging, so of course it prints what is theoretically needed to help debug.  There's nothing requiring you run with debug.

But anyhow if you run with debug, you will get the debug node format described in the documentation here which you will need to understand to interpret debug and tree files: https://github.com/verilator/verilator/blob/master/docs/internals.rst

The disables listed are because those are internally created signals, so not appropriate for tracing.

Closing as answered, but I suspect you have some larger question you didn't state which is why you are running with debug? Feel free to follow up here.

closed 2022-07-16 by wsnyder

burntfalafel, 2022-07-16:
@wsnyder thanks for your reply, my questions asked was more in the understanding of the trace. 

I am trying to build a coverage/profiler kind of tool which traces each of the variables in a HDL simulation. I assumed V3Coverage.cpp is the best place for me to start as it has the use of a [virtual `visit()` function ](https://github.com/verilator/verilator/blob/b0e796ca83cd05a6f72674c89e9af1eaa9c71366/src/V3Coverage.cpp#L384) which I could use to count a Verilog's `wire`, `reg`, etc. occurrence. I noticed the `nodep` variable being printed out [during the debug trace ](https://github.com/verilator/verilator/blob/b0e796ca83cd05a6f72674c89e9af1eaa9c71366/src/V3Coverage.cpp#L385)which is why I was asking a breakdown of it.

wsnyder, 2022-07-16:
You may want to see --stats output and also pull #3375.

burntfalafel, 2022-07-17:
Will do - thank you :)