verilator/verilator #3500: Paid development: support for fx68k

jotego, 2022-07-18:
The open source M68000 module [FX68K](https://github.com/ijor/fx68k) does not work on [Verilator](https://github.com/verilator/verilator/issues/3270). The problem seems to be the packed/unpacked structures (plus a couple of combinational optimization loops). The module author does not want to modify it to remove the structures. Commercial simulators and synthesis tools work correctly with his code. The open source simulators Verilator and Icarus verilog don't.

How much would it cost to pay a Verilator developer to add support to this module?

Thank you

udif, 2022-07-20:
By pure coincidence I took a look today at https://chipsalliance.github.io/sv-tests-results/?v=verilator+fx68k+fx68k and couldn't help noticing that fx68k is reported to be building fine under Verilator, with only warnings.
Since I remembered this issue, I couldn't help peeking into the report.

Looking into the warnings, I see this:
```
%Warning-UNPACKED: [third_party/cores/fx68k/fx68k.sv:48:9](https://chipsalliance.github.io/sv-tests-results/third_party/cores/fx68k/fx68k.sv.html#l-48): Unsupported: Unpacked struct/union
                                                                                : ... In instance fx68k
%Warning-UNPACKED: [third_party/cores/fx68k/fx68k.sv:73:9](https://chipsalliance.github.io/sv-tests-results/third_party/cores/fx68k/fx68k.sv.html#l-73): Unsupported: Unpacked struct/union
                                                                                : ... In instance fx68k
%Warning-UNPACKED: [third_party/cores/fx68k/fx68k.sv:56:9](https://chipsalliance.github.io/sv-tests-results/third_party/cores/fx68k/fx68k.sv.html#l-56): Unsupported: Unpacked struct/union
                                                                                : ... In instance fx68k
```
If this is unsupported, and the simulation will not run, why is this reported as a Warning and not an Error?

Besides, after reading IEEE1800-2017, I came to the conclusion that the only difference between packed and unpacked structs are:
1. You can treat packed structs as a superset of unpacked structs (i.e. there are several things you can do with a packed struct but not with an unpacked struct)
2. The only thing unpacked structs are providing over packed structs is that they can contain any other types.

So:
* if an unpacked struct contains only integral types or packed data types, why not simply implement it in the same way as a packed data type?
* If it contains types which cannot be used in a packed struct, then issue an error message, not a warning.

I took a look at the the other warnings out of curiosity, and noticed a few things:
1. fx68k is using `// synopsys translate on`  and `// synopsys translate off` which should really be avoided, and replaced with something along the lines of `` `ifdef SYNTHESIS`` ... `` `endif``
4. Verilator is not supporting `// synopsys translate on` on purpose. See https://github.com/verilator/verilator/issues/24
5. You can check my branch, https://github.com/udif/verilator/tree/synthesis_off , which I maintain for work-related issues, but please be aware that I merge it with the master branch only so-often, and if I were in your position, I would fix the code instead.

Given these 2 minor fx68k <> Verilator incompatibilities, I would simply fork a branch of fx68k and fix it if for my needs:
* Convert `// synopsys translate on`  / `// synopsys translate off` into `` `ifdef SYNTHESIS`` ... `` `endif``
* Convert the unpacked structs into packed ones by adding the `packed` keyword.

jotego, 2022-07-21:
Thank you for looking into this.

Verilator does not support mixed blocked and non-blocking assignemnts to a variable. This becomes a problem for the struct types because each signal within the struct is given a different treatment by the code, thus it is unpacked. There seem to be like 33 such errors. See the report below. I have no idea how they got the report you linked to.

I did try to remove the structs and use plain signals in my [fx68k branch](https://github.com/jotego/fx68k/tree/fix4), but although a formal tool seemed to be happy with the changes, Verilator still didn't simulate the result correctly so I guess my code wasn't really equivalent.

The author of fx68k is highly reluctant to making any changes to the code, particularly for something he argus is a simulator tool problem. My take at this is that it's better to design open source modules to work with FOSS tools, but well, open source comes with no warranties, so what can you do about it. 

```

%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:468:40: Operator COND expects 9 bits on the Conditional True, but Conditional True's CONST '4'h6' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  468 |    htemp = { 1'b0, binResult} + (lowC  ? 4'h6 : 4'h0);
      |                                        ^
                ... For warning description see https://verilator.org/warn/WIDTH?v=4.224
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:468:40: Operator COND expects 9 bits on the Conditional False, but Conditional False's CONST '4'h0' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  468 |    htemp = { 1'b0, binResult} + (lowC  ? 4'h6 : 4'h0);
      |                                        ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:469:32: Operator COND expects 5 bits on the Conditional True, but Conditional True's CONST '4'h6' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  469 |    hNib  = htemp[8:4] + (highC ? 4'h6 : 4'h0);
      |                                ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:469:32: Operator COND expects 5 bits on the Conditional False, but Conditional False's CONST '4'h0' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  469 |    hNib  = htemp[8:4] + (highC ? 4'h6 : 4'h0);
      |                                ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:473:40: Operator COND expects 9 bits on the Conditional True, but Conditional True's CONST '4'h6' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  473 |    htemp = { 1'b0, binResult} - (lowC  ? 4'h6 : 4'h0);
      |                                        ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:473:40: Operator COND expects 9 bits on the Conditional False, but Conditional False's CONST '4'h0' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  473 |    htemp = { 1'b0, binResult} - (lowC  ? 4'h6 : 4'h0);
      |                                        ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:474:32: Operator COND expects 5 bits on the Conditional True, but Conditional True's CONST '4'h6' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  474 |    hNib  = htemp[8:4] - (highC ? 4'h6 : 4'h0);
      |                                ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68kAlu.sv:474:32: Operator COND expects 5 bits on the Conditional False, but Conditional False's CONST '4'h0' generates 4 bits.
                                                            : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.alu.aluCorf
  474 |    hNib  = htemp[8:4] - (highC ? 4'h6 : 4'h0);
      |                                ^
%Warning-WIDTH: ../../../../modules/fx68k/fx68k.sv:1930:19: Operator ASSIGN expects 4 bits on the Assign RHS, but Assign RHS's VARREF 'i' generates 32 bits.
                                                          : ... In instance game_test.u_game.u_sub.u_cpu.u_cpu.excUnit.rmPren
 1930 |              hbit = i;
      |                   ^
%Warning-UNPACKED: ../../../../modules/fx68k/fx68k.sv:48:9: Unsupported: Unpacked struct/union
                                                          : ... In instance game_test.u_game
   48 | typedef struct {
      |         ^~~~~~
%Warning-UNPACKED: ../../../../modules/fx68k/fx68k.sv:73:9: Unsupported: Unpacked struct/union
                                                          : ... In instance game_test.u_game
   73 | typedef struct {
      |         ^~~~~~
%Warning-UNPACKED: ../../../../modules/fx68k/fx68k.sv:56:9: Unsupported: Unpacked struct/union
                                                          : ... In instance game_test.u_game
   56 | typedef struct {
      |         ^~~~~~
[... cut ...]
                                                             : ... In instance game_test.u_game.u_main.u_cpu
  293 |  s_nanod Nanod;
      |          ^~~~~
                    ../../../../modules/fx68k/fx68k.sv:293:10:      Example path: game_test.u_game.u_main.u_cpu.u_cpu.Nanod
                    ../../../../modules/fx68k/fx68k.sv:790:22:      Example path: ASSIGNW
                    ../../../../modules/fx68k/fx68k.sv:293:10:      Example path: game_test.u_game.u_main.u_cpu.u_cpu.Nanod
%Error: Exiting due to 33 error(s)
```