verilator/verilator #3510: Initializing string in struct along with numeric value throw Internal Error

MG525, 2022-07-21:
The issue was encountered when I was working on fixing [Support pattern member assignment with data type #2328](https://github.com/verilator/verilator/issues/2328 ) and created this sample Test Case from IEEE 1800-2017 SV CH:10.9.2

**Example:**
```
module top();
    typedef struct {
        logic [7:0] a;
        string s;
      } test;
    test x;
    initial begin;
        x = '{a: '1, s: "test"};
        $display(x.s);
        $finish;
    end
endmodule
```

**Error:**
```
%Error: Internal Error: ../V3Number.cpp:2392: Number operation called with non-string argument: '8'hff"
                                                                                    : ... In instance top
x = '{a: '1, s: "test"};
             ^
```

**Version:**
_git master:_ _Verilator 4.225 devel rev v4.224-49-ge0b61cea_

**Notes:**
The issue "Internal error on string parameter in latest master #3215" seems similar but the fix did not hit this one

Any information on what might be happening would be much appreciated

MG525, 2022-08-08:
I investigated the issue for some time and the issue might not be related to string but it is a general issue where Struct node when replaced with AstConcat node then verilator is expecting integr_type when there is concatenation.

If "string" is replaced by "real" in the above example then we got error below:

```
   12 |         x = '{s: 10.01, a: '1};
      |                  ^~~~~
                  ... For warning description see https://verilator.org/warn/REALCVT?v=4.225
                  ... Use "/* verilator lint_off REALCVT */" and lint_on around source to disable this message.
dot -Tpdf -o ~/a.pdf obj_dir/Vtop_051_task_call.dot
%Error: Internal Error: /home/mg/verilator/my_repos/verilator/examples/my_example/top.v:15:20: ../V3Number.cpp:871: Real operation on wrong sized/non-real number
   15 |         $display(x.s);
```

Also looking back at the string issue I noticed that "visit AstConcat" when RHS or LHS is a string then it is replacing AstConcat with AstConcatN where in this case we are expecting both inputs are strings as far as I undertood from VNumber operations

wsnyder, 2022-08-08:
These are side effects of #1541, and likely can't be fixed until that one is.

closed 2023-01-29 by wsnyder