verilator/verilator #3511: Use of the default keyword applies to members in nested structures not working as expected

MG525, 2022-07-22:
The issue was encountered when I was working on fixing [Support pattern member assignment with data type #2328](https://github.com/verilator/verilator/issues/2328 ) and created this sample Test Case from IEEE 1800-2017 SV CH:10.9.2

The standard mentioned that "Use of the default keyword applies to members in nested structures or elements in unpacked arrays in
structures" but this is not the case as below

**Example:**
```
module top();
/* verilator lint_off UNPACKED */

  struct { // IEEE 1800-2017 SV CH:10.9.2
    int A;
    struct {
    int B, C;
    } BC1, BC2;
  } ABC, DEF;

  
  initial begin;

  DEF = '{default:10}; 
  $display("DEF.A=%d, DEF.BC1.B=%d, DEF.BC1.C=%d, DEF.BC2.B=%d, DEF.BC2.C=%d", DEF.A, DEF.BC1.B, DEF.BC1.C, DEF.BC2.B, DEF.BC2.C);

  $finish;
  end

endmodule
```

**Output:**

`DEF.A=         10, DEF.BC1.B=          0, DEF.BC1.C=         10, DEF.BC2.B=          0, DEF.BC2.C=         10`

**Error:**
_Value of 10 is not applied to DEF.BC1.B and DEF.BC2.B
They are assigned 0_

**Version:**
_git master:_ _Verilator 4.225 devel rev v4.224-49-ge0b61cea_

**Notes:**

Any information on what might be happening would be much appreciated

wsnyder, 2022-07-22:
Agreed, this should work. V3Width.cpp patternUOrStruct is what applies the defaults, Perhaps you could figure out and make a pull that recursively apply the default on structs under structs?

MG525, 2022-07-30:
Hi @wsnyder 

Now the issue is when struct member is not initialized it took the value of default which is "10" in the example so C is assigned with "10" and B is assigned "0"

To solve this I have two options on my mind:
1. Iterate with a nested for loop whenever we hit struct not yet initialized (might be easy but needs  refactoring for the code) .. Also we need to keep in mind that "type_default" also to be considered as mentioned by the SV standard not only "default"

2. Replace the pattern with the value "10" with a new pattern with value "`{default: 10}" which in my first trial to create a new AstPatMember with this specifications I hit an internal error **"Pattern member type not assigned by AstPattern visitor" (W3Width.cpp line: 3904)**

So would the implementation of the 2nd option anyway applicable? If so then any hints for this implementation would be much appreciated as I am stuck at the internal error

wsnyder, 2022-07-30:
Using another simulator can you update the above example with what all the expected values are?  That is are the substructure fields all 10 (assuming it fits in the width of the subfield).

wsnyder, 2022-07-30:
I suspect method 1 will work better, though agree it'll need a bit more refactoring.

MG525, 2022-07-30:
> 

Using edaplayground with "Aldec Riviera Pro 2022.04" with the enhanced version of the example:

```
// Code your testbench here
// or browse Examples
module top();
/* verilator lint_off UNPACKED */

  struct { // IEEE 1800-2017 SV CH:10.9.2
    int A;
    struct {
    int B, C;
    } BC1, BC2;
  } ABC, DEF;

  
  initial begin;

  DEF = '{default:10}; 
  $display("DEF.A=%d, DEF.BC1.B=%d, DEF.BC1.C=%d, DEF.BC2.B=%d, DEF.BC2.C=%d", DEF.A, DEF.BC1.B, DEF.BC1.C, DEF.BC2.B, DEF.BC2.C);
    
    DEF = '{int:5}; 
  $display("DEF.A=%d, DEF.BC1.B=%d, DEF.BC1.C=%d, DEF.BC2.B=%d, DEF.BC2.C=%d", DEF.A, DEF.BC1.B, DEF.BC1.C, DEF.BC2.B, DEF.BC2.C);
    
    DEF = '{int:5, default: 10}; 
  $display("DEF.A=%d, DEF.BC1.B=%d, DEF.BC1.C=%d, DEF.BC2.B=%d, DEF.BC2.C=%d", DEF.A, DEF.BC1.B, DEF.BC1.C, DEF.BC2.B, DEF.BC2.C);

  $finish;
  end

endmodule
```

**_Result:_**
```

# KERNEL: DEF.A=         10, DEF.BC1.B=         10, DEF.BC1.C=         10, DEF.BC2.B=         10, DEF.BC2.C=         10
# KERNEL: DEF.A=          5, DEF.BC1.B=          5, DEF.BC1.C=          5, DEF.BC2.B=          5, DEF.BC2.C=          5
# KERNEL: DEF.A=          5, DEF.BC1.B=          5, DEF.BC1.C=          5, DEF.BC2.B=          5, DEF.BC2.C=          5
# RUNTIME: Info: RUNTIME_0068 testbench.sv (25): $finish called.
```

wsnyder, 2022-08-12:
Thanks for fixing this!

PR #3524 (2022-08-12): Fix #3511: Nested default assignment for struct pattern
Fix #3511

1. The new implementation guarantees default assignment using "default" or "type_default" in nested struct
2. Some refactoring has been done to reuse the code in different phases
3. Recursion is used when struct member is hit as a member and not yet matched
4. Modified the test case to hit the new implementation
5. Removed legacy comment in bison parser
closed 2022-08-12 by wsnyder