verilator/verilator #3514: Formatting in docs/guide/files.rst

dbear496, 2022-07-23:
In [docs/guide/files.rst](https://github.com/verilator/verilator/blob/e0b61ceabd76d2fd3aae0a1ff1bd522e307de409/docs/guide/files.rst), there seem to be some stray asterisks in the 'Files Read/Written' section, e.g. `{prefix}*___024root*{__n}.cpp` looks like it has some un-rendered markup. Also, `{prefix|_hierMkArgs.f` appears to have a pipe instead of a close brace.

Additionally, I noticed that the actual output files I see do not match exactly to the table. For example, I got an output file named `VMyModule___024root__DepSet_h59f7838a__9.cpp`, and I don't know what the `DepSet_h59f7838a` part means.

verilator --version: Verilator 4.220 2022-03-12 rev v4.220-23-ged979c6a (mod)
OS: Ubuntu 22.04

gezalore, 2022-07-23:
The `*` are wildcards as you suspect (to explain in docs). The DepSet hash depends on which headers the file includes. This is intended to help with incremental re-compilation of the output via ccache but is otherwise unimportant.

dbear496, 2022-07-23:
Okay, thanks for clarifying the purpose of the DepSet hash and that the asterisks are meant to be there as wildcards. My question is answered, but I will just leave this issue open in case any of the other formatting still needs to be fixed. To clarify, here are the rows that I still think look suspicious:

| file | sus |
| --- | --- |
| *{prefix\|*_hierMkArgs.f | pipe instead of close brace |
| *{prefix}*___024root__Trace{__n}\*.cpp | {__n} is missing italics and asterisk may be extra |
| *{prefix}*___024root__Trace__Slow{__n}\*.cpp | {__n} is missing italics and asterisk may be extra |

If these are actually the way they are supposed to be, then feel free to close the issue.

closed 2022-07-28 by wsnyder